micro-manager/micro-manager #1000: Circular buffer size limit too small

beeeel, 2020-11-08:
The circular buffer size limit is hardcoded in MMCore/CircularBuffer.cpp:

`const unsigned long maxCBSize = 100000;    //a reasonable limit to circular buffer size

// set a reasonable limit to circular buffer capacity 
if (cbSize > maxCBSize)
     cbSize = maxCBSize;`

I have an experiment where I wish to collect >1 million small images as quickly as my camera allows (64x64px, ~1600fps), in order to measure Brownian motion of a constrained particle. 

Is this limit actually needed? If not, could it be removed? Otherwise, would it be possible to change it to 10,000,000 instead of 100,000?

marktsuchida, 2020-11-10:
The reason for this limit appears to be lost in history (without some very deep digging, at least). I think we can increase it if it can be confirmed that this limit is not masking some other bug (e.g. `frameSizeBytes` several lines above being zero), and we make sure to check for integer overflow (esp. when computing `cbSize`).

Every time in the past that I've encountered much more than 1k fps sustained, I've concluded that Micro-Manager is not equipped to deal with it: among other things, the metadata handling is too slow. But I guess storing the entire acquisition in the sequence buffer (instead of its intended use as a transitive buffer) might be a valid workaround in some cases. Does your acquisition work okay when the buffer size is 100k images (which should last 1 min even if no images are being popped)? I'm curious how long it takes to retrieve the images.

marktsuchida, 2020-11-10:
Related: https://forum.image.sc/t/real-time-processing/44410/9

marktsuchida, 2020-11-10:
I went ahead and increased the limit to 10M, because (being much smaller than the `unsigned long` maximum) it should be safe.

closed 2020-11-10 by marktsuchida

beeeel, 2020-11-12:
Thanks for making this change! 

Regarding the acquisition, if I run it with 100k images, and only pop them after the acquisition is finished, it takes ~60s to acquire and ~50s to pop 100k tagged images. Like you say, the metadata handling is slow. If I pop the pixel data only (mmc.popNextImage()), it takes 0.9s to pop 100k images. 

Since my experiment doesn't benefit much from saving the metadata, I intend to save it only for the first and last frames, or possibly for 1 in every 10k frames.

nicost, 2020-11-12:
Nice!  So, that suggests that - in your case - metadata handling is about 0.5 ms per image.  I am wondering if it would make sense to add in a third variant to pop an image from the Circular Buffer.  This one would not copy the complete state of the system, but only metadata that were inserted by the camera itself (which can have useful things like hardware timestamps, etc..).  Seems something worthy to investigate.

PR #1002 (2020-11-10): Increase sequence buffer size (frame count) limit
Closes #1000.