micro-manager/micro-manager #1001: System error when developing the plugins

bryanchang0603, 2020-11-10:
When I develops the plugin for twisting the GUI, and moved the image configration window to the main frame, I encountered this error. 
org.micromanager.display.internal.displaywindow.imagej.MMImageCanvas.access$100(MMImageCanvas.java:34)
Can someone help me to solve this problem? thank you su much!

nicost, 2020-11-10:
Can you post the complete stack trace?

marktsuchida, 2020-11-10:
Can you also explain what you mean by "moved the image configuration window to the main frame"?

bryanchang0603, 2020-11-10:
here is the complete stack trace
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at org.micromanager.display.internal.displaywindow.imagej.ImageJBridge.ij2mmRoiMayHaveChanged(ImageJBridge.java:897)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at org.micromanager.display.internal.displaywindow.imagej.MMImageCanvas.repaint(MMImageCanvas.java:378)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at org.micromanager.display.internal.displaywindow.imagej.MMImageCanvas.handleComponentResized(MMImageCanvas.java:173)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at org.micromanager.display.internal.displaywindow.imagej.MMImageCanvas.handleHierarchyChanged(MMImageCanvas.java:180)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at org.micromanager.display.internal.displaywindow.imagej.MMImageCanvas.access$100(MMImageCanvas.java:34)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at org.micromanager.display.internal.displaywindow.imagej.MMImageCanvas$2.hierarchyChanged(MMImageCanvas.java:61) After this are non MM error
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.processHierarchyEvent(Component.java:6706)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.processEvent(Component.java:6325)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.dispatchEventImpl(Component.java:4889)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.dispatchEvent(Component.java:4711)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.createHierarchyEvents(Component.java:5555)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Container.createHierarchyEvents(Container.java:1444)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Container.createHierarchyEvents(Container.java:1444)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Container.createHierarchyEvents(Container.java:1444)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Container.removeAll(Container.java:1316)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at org.micromanager.quickaccess.internal.controls.SnapButton$5.actionPerformed(SnapButton.java:160)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.processMouseEvent(Component.java:6539)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.processEvent(Component.java:6304)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Container.processEvent(Container.java:2239)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.dispatchEventImpl(Component.java:4889)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Container.dispatchEventImpl(Container.java:2297)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.dispatchEvent(Component.java:4711)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Container.dispatchEventImpl(Container.java:2283)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Window.dispatchEventImpl(Window.java:2746)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.Component.dispatchEvent(Component.java:4711)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventQueue.access$500(EventQueue.java:97)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventQueue$3.run(EventQueue.java:709)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventQueue$3.run(EventQueue.java:703)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.security.AccessController.doPrivileged(Native Method)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventQueue$4.run(EventQueue.java:733)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventQueue$4.run(EventQueue.java:731)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.security.AccessController.doPrivileged(Native Method)
[... cut ...]
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2020-11-09T23:50:25.572294 tid20188 [IFO,App]   at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

bryanchang0603, 2020-11-10:
For moved to the main frame, I modified the functions in the SnapeLiveManager, the InspectorController, the Snapbutton and other files to allow InspectorController is able to pass the configration jpanel and added to the main frame. I am not sure about how the other classes are interacting with the configration window as if I stop creating it, the configration panel will not be generated. So I left it minimized, but no configration panel is in there.

nicost, 2020-11-10:
and what is the actual Exception (i.e., the line just above the stack trace)?

bryanchang0603, 2020-11-10:
its null pointer exception. so in this case, but I am not sure what is missing here.

nicost, 2020-11-11:
Line 897 of ImageJBridge has:

` Roi roi = imagePlus_.getRoi();`

The only thing that can be null there is `imagePlus_`.  It is set in the initialize function on line 188 (`imagePlus_ = MMImagePlus.create(this);`).  You can set a breakpoint there to make sure that it the initialize function is called in your code, and that that line returns a non-null object.  Also, code on line 235 (that deals with window closing), will set imagePlus_ back to null. You could always add a null check before line 897, but I have the feeling that will not prevent a similar error popping up elsewhere.

bryanchang0603, 2020-11-11:
Got it. Thank you so much. I will try this now.

nicost, 2020-11-13:
OK to close?

bryanchang0603, 2020-11-14:
yeah, thank you so much!

closed 2020-11-14 by nicost