micro-manager/micro-manager #1006: error at build time

oopsShouldntHaveDoneThat, 2020-11-15:
I get an error at build time on linux. I have used both the github code and the svn mirror. Results are practically the same. It happens both when i try to install natively or as an ImageJ plugin. This is what I get when I run the 'make' command.

`Making install in buildscripts/AntExtensions
make[1]: Entering directory '/home/myuseraccount/.mm/micro-manager/buildscripts/AntExtensions'
JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64 /usr/bin/ant -Dmm.javacflags="-Xlint:all,-path,-serial" -logger org.apache.tools.ant.listener.SimpleBigProjectLogger jar
Buildfile: /home/myuseraccount/.mm/micro-manager/buildscripts/AntExtensions/build.xml

BUILD SUCCESSFUL
Total time: 0 seconds
make[2]: Entering directory '/home/myuseraccount/.mm/micro-manager/buildscripts/AntExtensions'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/myuseraccount/.mm/micro-manager/buildscripts/AntExtensions'
make[1]: Leaving directory '/home/myuseraccount/.mm/micro-manager/buildscripts/AntExtensions'
Making install in MMDevice
make[1]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMDevice'
Making install in .
make[2]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMDevice'
make[3]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMDevice'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMDevice'
make[2]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMDevice'
Making install in unittest
make[2]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMDevice/unittest'
make[3]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMDevice/unittest'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMDevice/unittest'
make[2]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMDevice/unittest'
make[1]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMDevice'
Making install in MMCore
make[1]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMCore'
Making install in .
make[2]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMCore'
make[3]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMCore'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMCore'
make[2]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMCore'
make[1]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMCore'
Making install in MMCoreJ_wrap
make[1]: Entering directory '/home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap'
JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64 /usr/bin/ant -Dmm.javacflags="-Xlint:all,-path,-serial" -logger org.apache.tools.ant.listener.SimpleBigProjectLogger -Dsrcdir=gensrc jar
Buildfile: /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/build.xml

MMCoreJ.compile:
 [mm-javac] Compiling 29 source files to /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/build
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:231: error: incompatible types: Long cannot be converted to int
 [mm-javac]          Rectangle r = new Rectangle((int) xs.get(i), (int) ys.get(i),
 [mm-javac]                                                  ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:231: error: incompatible types: Long cannot be converted to int
 [mm-javac]          Rectangle r = new Rectangle((int) xs.get(i), (int) ys.get(i),
 [mm-javac]                                                                   ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:232: error: incompatible types: Long cannot be converted to int
 [mm-javac]                (int) widths.get(i), (int) heights.get(i));
 [mm-javac]                                ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:232: error: incompatible types: Long cannot be converted to int
 [mm-javac]                (int) widths.get(i), (int) heights.get(i));
 [mm-javac]                                                      ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:248: error: incompatible types: int cannot be converted to Long
 [mm-javac]          xs.add(r.x);
 [mm-javac]                  ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:249: error: incompatible types: int cannot be converted to Long
 [mm-javac]          ys.add(r.y);
 [mm-javac]                  ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:250: error: incompatible types: int cannot be converted to Long
 [mm-javac]          widths.add(r.width);
 [mm-javac]                      ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:251: error: incompatible types: int cannot be converted to Long
 [mm-javac]          heights.add(r.height);
 [mm-javac]                       ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/StrVector.java:73: warning: [cast] redundant cast to int
 [mm-javac] 			String strs[] = new String[(int) size()];
 [mm-javac] 			                           ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/LongVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Integer ints[] = new Integer[(int) size()];
 [mm-javac]                                       ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/DoubleVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Double ints[] = new Double[(int) size()];
 [mm-javac]                                     ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/UnsignedVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Long ints[] = new Long[(int) size()];
 [mm-javac]                                 ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CharVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Character ints[] = new Character[(int) size()];
 [mm-javac]                                           ^
 [mm-javac] /home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/StrMap.java:66: warning: [cast] redundant cast to String
 [mm-javac]     Iterator itr = find((String) key);
 [mm-javac]                         ^
 [mm-javac] Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
 [mm-javac] 8 errors
 [mm-javac] 7 warnings

BUILD FAILED
/home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap/build.xml:15: Compile failed; see the compiler error output for details.

Total time: 1 second
make[1]: *** [Makefile:816: MMCoreJ.jar] Error 1
make[1]: Leaving directory '/home/myuseraccount/.mm/micro-manager/MMCoreJ_wrap'
make: *** [Makefile:521: install-recursive] Error 1
`

nicost, 2020-11-17:
Somehow reminds of previous issues building MM with Swig 4.0 (#837, #838, #931).  Try using Swig 3.0 instead.  Also, we build MM with Java 8. Java 11 may or may not work.

oopsShouldntHaveDoneThat, 2020-11-17:
> Somehow reminds of previous issues building MM with Swig 4.0 (#837, #838, #931). Try using Swig 3.0 instead. Also, we build MM with Java 8. Java 11 may or may not work.

Downgrading swig to 3.0 fixed the problem. Java 11 seems to work fine for me.

nicost, 2020-11-17:
Thanks for letting us know!

closed 2020-11-17 by nicost