micro-manager/micro-manager #1016: Clarify policy on when summary metadata of datastore can be set

marktsuchida, 2020-11-18:
- Docs say can only be set once (good!)
- We probably need to also enforce only setting it before adding images
  - Except for rewritable datastores, of course
  - Does this restriction work, though?
  - See also `Pipeline.insertSummaryMetadata()` Javadoc

nicost, 2021-04-09:
Note this is related to parts #1017 since most Storage methods rely on the summary metadata to "know" which axes they contain (or eventually will contain), so addressing remaining issues with Axes in DataProviders (and Datastores) need to know whether and how they can rely on SummaryMetadata.

marktsuchida, 2021-04-28:
@nicost I though that made sense, but now I'm confused: what exactly do you mean by remaining issues with axes?

closed 2021-04-28 by nicost

marktsuchida, 2021-04-28:
(See #1210 regarding the "remaining issues with Axes".)

PR #1209 (2021-04-28): Forbid setting summary metadata after adding images to datastore
Resolves #1016. As best as I can tell, there is no existing code that tries to add images before adding metadata.