micro-manager/micro-manager #1017: Coords axes policy needs clarification

marktsuchida, 2020-11-18:
- When (if ever) do zeros get added for TPZC; can we stop doing so?
- Where are non-TPZC axes allowed (everywhere except file-backed Datastores?)
- Are axes ordered or not? Are comparison operations etc. well-defined and self-consistent?
- How are missing axes treated in `DataProvider` and `Datastore` methods?

See also #619.

nicost, 2020-11-28:
> When (if ever) do zeros get added for TPZC; can we stop doing so?

Not sure what this means.  For each and every instance of Coords all four prime axes need to be set to 0 by the calling code, or bad things will happen (such as not displaying the image with that Coord when added to a Datastore) as described in #619.  Not adding zeroes is currently not an option, but it would be great if the code would be fixed so that it would be possible.  What is the planned approach to improve on the current code?

marktsuchida, 2020-11-30:
Precisely. It is clear that in the long term, everything should just work without having to manually add zeroes. For that, the issues in #619 (and possibly others) need to be solved. If that is not easy in the short term (and therefore we need to keep the rule of users always having to add the 4 axes), at the very least we need a transition plan so that the restriction can be lifted in the future without breaking code written under the current restrictions. So either fixing #619 (very unlikely) or clearly documenting the current rules and future plans is what this issue is about.

This may not just be about the datastore code, but _might_ potentially involve the mathematical model of what a `Coords` is: for example, the definition of equality between two `Coords` values. Or maybe we need to introduce a new comparison operation ("are these two `Coords` equal given these particular axes?"). Alternatively, it might be as simple as documenting limitations of `Datastore` (rather than `Coords` in general). I intend to find the simplest future-proof thing we can do for the moment, but leaving these issues vague seems like a bad idea.

nicost, 2020-11-30:
Great!  I like the idea of introducing a comparison operator.  Maybe it is as simple as dropping all zero axes, and only then testing for equality.

marktsuchida, 2021-04-28:
The remaining issue has been restated as #1210.

closed 2021-04-28 by marktsuchida