micro-manager/micro-manager #1019: Builder APIs should be standardized

marktsuchida, 2020-11-18:
- The builder for `T` should be a nested interface `T.Builder` with a method `build()`.
- The method of `T` to obtain a copy builder should be named `copyBuilder()`, not `copy()` etc.
- Generic builder factory methods should be named `builder()` (if it is obvious for which type) or `tBuilder()`. Not `getTBuilder()`, `newTBuilder()` etc.
- Changes to conform to this rule should keep existing methods but deprecate them.

nicost, 2020-11-30:
In addition, I highly prefer there to be instance methods to obtain a Builder, rather than a static method (or only a static method).  Reason for this is that the static methods make scripting harder.  At the very least, and import statement is needed, and it is quite difficult to discover where these are and what is needed, whereas an instance method makes an import unnecessary, and can be discovered easily through the documentation.  Example: "CoordsBuilder".  The deprecated method `cb = mm.data().getCoordsBuilder()` is much easier to work with than the: `import org.micromanager.data.Coordinates; cb = Coordinates.Builder();`

marktsuchida, 2020-11-30:
Sounds good; the argument for scripting fluency is quite strong. I'll add `mm.data().coordsBuilder()` (keeping the name with `get` deprecated). Is this the only builder in need of a nonstatic method?

nicost, 2020-12-01:
A not-thoroughly researched list:
- SequenceSettings
- PropertyMap

PR #1195 (2021-04-17): Make all Builders uniform
Closes #1019

Only thing missing is access function for PropertyMaps.Builder().  I am not sure where that function should be.  The only logical place is the Studio interface itself. Suggestions?

PropertyMaps is a bit chaotic with the Builder extending a deprecated interface. Would be nice to clean this all up after the release.
closed 2021-04-17 by nicost