micro-manager/micro-manager #1023: MultiStagePosition has a static function with a MultiStagePosition instance as an argument

nicost, 2020-11-18:
org.micromanager.MultiStagePosition contains the static function:

`public static void goToPosition(MultiStagePosition msp, CMMCore core_) throws Exception`

It seems strange not use:

`public void goToPosition(CMMCore core) throws Exception `

instead.

In addition, the function is inefficient as it moves each stage, waits for it to stop moving, and then moves the other.  Also, the `core_` argument should be `core` according to the style guide.

marktsuchida, 2020-11-18:
Agreed. This should be removed from `MultiStagePosition` entirely if possible, making the class not dependent on MMCore and closer to a clean pure-data object. Not yet sure exactly where to place it, although it might be good to consider a stage control utility class.

Also agree that movements should be fully parallel.

The method is currently used by the GUI actions to move the stage, in the Position List and Create Grid dialogs.

nicost, 2020-11-18:
Have a look at PR#1025, it may not be completely what you have in mind, but a step in the right direction, and would allow our GUI to call this better implementation.  Or we can wait until you have a good idea where the goToPosition method should move.

marktsuchida, 2020-11-18:
Yeah, sorry I commented before seeing #1025; also added a comment there.