micro-manager/micro-manager #1036: OSX Security problems that prevent MMStudio from opening

henrypinkard, 2020-12-06:
For some time now, when downloading a new nightly build on OSX, Micro-Manager does not open when opening ImageJ, until the application is dragged to the desktop and back as described [here](https://micro-manager.org/wiki/Download_Micro-Manager_Latest_Release). Even though these instructions are up, this seems to be a source of consistent confusion for new users and it would really make getting started with MM a lot easier if it were fixed

nicost, 2020-12-06:
That would be wonderful, but I do believe that the only one who could fix this is Apple.  If there are any other ways, I am all ears (and the whole ImageJ community using Macs will be thankful;).

There are many more issues associated with the installers.  Both the Windows and Mac installers should be signed to avoid prompts that the application can not be trusted.  There also are reports that on the latest MacOS each loaded device adapter (.so file) causes a warning message.  These should all be high priority tickets.

Pinaceae, 2023-01-11:
I just tried the latest nightly build. Unlike MM 2.0.0, it did throw out some error messages-

[image]

marktsuchida, 2024-08-01:
See #2001.

closed 2024-08-01 by marktsuchida