micro-manager/micro-manager #1051: Beanshell sequence acquisition issue with NI100x-AnalogIO-based "DA Z stage" focus device

linshaova, 2020-12-11:
Hello there,

(OS: Windows 10 build 1909; MM version: 2.0 gamma1 20201130. Thought to be and reported as a [pycro-manager issue](https://github.com/micro-manager/pycro-manager/issues/194) but it turns out not to be.)

I have a voltage-driven focus device controlled by a NI DAQ's AO port and interfaced to MM via the "DA Z Stage" and NI100x drivers. I wanted to mimic a simple MDA acquisition using the following script (no image is saved; just to see if the hardware works):
```java
import mmcorej.DoubleVector;

// Create a vector for uploading a sequence of z stage positions:
DoubleVector zSequence = new DoubleVector();
int nZ = 20;
for (i=0; i<nZ; i++)
    zSequence.add(i); // i um
// Load the sequence
core_ = mm.core();
zStage = core_.getFocusDevice();
core_.loadStageSequence(zStage, zSequence);
core_.prepareSequenceAcquisition(core_.getCameraDevice());
core_.startStageSequence(zStage);

core_.startSequenceAcquisition(nZ, 0, true);

while (core_.isSequenceRunning()) {
    //wait
}

core_.stopSequenceAcquisition();
core_.stopStageSequence(zStage); // without this, "loadStageSequence" would just hang in the next run
```
The first run of this seems fine, but the subsequent runs throw an exception `Error calling WriteAnalogF64: Task specified is invalid or does not exist` at the line `core_.loadStageSequence(zStage, zSequence)`:
```
2020-12-11T15:26:42.608002 tid7908 [IFO,dev:AO1] Error calling WriteAnalogF64: Task specified is invalid or does not exist.
                                   [           ] Status Code: -200088
2020-12-11T15:26:42.610003 tid7908 [IFO,App] Error: Line 11: run-time error : Error in device "Galvo": (Error message unavailable) (-200088)
```
(my focus device is referred to as "Galvo"; sorry for the confusion!)

And if the last line `core_.stopStageSequence(zStage)` was not present in the first run, then the run after that would just free MM and it needs force-quit.

Does this indicate something wrong with the script or potential mishandling of NI devices by MM? Thank you very much!

lin

linshaova, 2020-12-11:
I forgot to mention that MDA works fine with the same hardware config. Here is the relevant part of the config file:
```
# Generated by Configurator on Tue Dec 01 10:30:03 EST 2020

# Reset
Property,Core,Initialize,0

# Devices
Device,Zyla,AndorSDK3,Andor sCMOS Camera
Device,AO1,NI100X,AnalogIO
Device,Galvo,Utilities,DA Z Stage
Device,yStage,ThorlabsAPTStage,SCC001
Device,xStage,ThorlabsAPTStage,SCC001
Device,zStage,ThorlabsAPTStage,TDC001
Device,Combo XY Stage,Utilities,Combo XY Stage
Device,dio_port1,NI100X,DigitalIO
Device,TTLshutter1,Utilities,State Device Shutter

# Pre-init settings for devices
Property,AO1,AutoDetectedOutputPort,Dev1/ao1
Property,AO1,Demo,No
Property,AO1,IOChannel,Dev1/ao1
Property,AO1,MaxVolts,5.0000
Property,AO1,MinVolts,-5.0000
Property,AO1,TriggerSequenceLength,10000
Property,Galvo,Stage High Position(um),945.0000
Property,Galvo,Stage High Voltage,3.0000
Property,Galvo,Stage Low Position(um),-945.0000
Property,Galvo,Stage Low Voltage,-3.0000
Property,yStage,Channel,1
Property,yStage,Position Lower Limit (um),0.0000
Property,yStage,Position Upper Limit (um),100000.0000
Property,yStage,Serial Number,90993081
Property,xStage,Channel,1
Property,xStage,Position Lower Limit (um),0.0000
Property,xStage,Position Upper Limit (um),100000.0000
Property,xStage,Serial Number,90993082
Property,zStage,Channel,1
Property,zStage,Position Lower Limit (um),0.0000
Property,zStage,Position Upper Limit (um),12000.0000
Property,zStage,Serial Number,27505731
Property,Combo XY Stage,SimulatedXStepSizeUm,0.0100
Property,Combo XY Stage,SimulatedYStepSizeUm,0.0100
Property,dio_port1,AutoDetectedOutputPort,Dev1/port1
Property,dio_port1,IOChannel,Dev1/port1
Property,dio_port1,TriggerSequenceLength,1000

# Pre-init settings for COM ports

# Hub (parent) references

# Initialize
Property,Core,Initialize,1

# Delays

# Focus directions
FocusDirection,Galvo,0
FocusDirection,yStage,0
FocusDirection,xStage,0
FocusDirection,zStage,-1

# Roles
Property,Core,Camera,Zyla
Property,Core,Shutter,TTLshutter1
Property,Core,Focus,Galvo
Property,Core,AutoShutter,1

# Camera-synchronized devices

# Labels
# dio_port1
Label,dio_port1,255,255
Label,dio_port1,254,254
Label,dio_port1,253,253
......
Label,dio_port1,4,laser488
Label,dio_port1,3,3
Label,dio_port1,2,2
Label,dio_port1,1,1
Label,dio_port1,0,closed

# Configuration presets
# Group: cameraROI
# Preset: 1392x1040
ConfigGroup,cameraROI,1392x1040,Zyla,AcquisitionWindow,1392x1040

# Preset: Full Image
ConfigGroup,cameraROI,Full Image,Zyla,AcquisitionWindow,Full Image

# Preset: 1024x1024
ConfigGroup,cameraROI,1024x1024,Zyla,AcquisitionWindow,1024x1024
......

# Group: Channel
# Preset: 560
ConfigGroup,Channel,560,dio_port1,State,8

# Preset: 488
ConfigGroup,Channel,488,dio_port1,State,4

# Group: System
# Preset: Startup
ConfigGroup,System,Startup,Zyla,AuxiliaryOutSource (TTL I/O),FireRow1
ConfigGroup,System,Startup,AO1,TriggerInputLine,/Dev1/PFI8
ConfigGroup,System,Startup,dio_port1,TriggerInputLine,/Dev1/PFI8
ConfigGroup,System,Startup,Galvo,Position,0.0000
ConfigGroup,System,Startup,yStage,Velocity (mm/s),10.0000
ConfigGroup,System,Startup,xStage,Velocity (mm/s),10.0000
ConfigGroup,System,Startup,zStage,Velocity (mm/s),1.0000
ConfigGroup,System,Startup,Combo XY Stage,PhysicalStage-X,xStage
ConfigGroup,System,Startup,Combo XY Stage,PhysicalStage-Y,yStage
ConfigGroup,System,Startup,TTLshutter1,State Device,dio_port1
ConfigGroup,System,Startup,Zyla,AcquisitionWindow,1920x1080
ConfigGroup,System,Startup,dio_port1,State,4

# Group: Galvo Position
# Preset: NewPreset
ConfigGroup,Galvo Position,NewPreset,Galvo,Position,0.0000

# PixelSize settings
# Resolution preset: TL200mm
ConfigPixelSize,TL200mm,Core,Camera,Zyla
PixelSize_um,TL200mm,0.2468
PixelSizeAffine,TL200mm,1.0,0.0,0.0,0.0,1.0,0.0

```

Thanks!

nicost, 2021-01-16:
Sorry for the delay.  I think that your MDA does not use a sequence for the NI100x.  Can you use the NIMultiAnalog device adapter instead?  Mark wrote that to replace the NI100x, so possibly it will do things the way you would like them to be.

nicost, 2022-07-17:
The NI100x device adapter has been deprecated and fully replaced with the NIDAQ adapter.  Can you test that one and re-open the issue if this is still a problem?

closed 2022-07-17 by nicost