micro-manager/micro-manager #1060: Newer FLIR/Spinnaker SDK support

rguerrab, 2020-12-22:
Hello,

Would it be possible to update the version of the FLIR SDK that is supported by micro-manager?  Micro-manager fills a nice slot for quick alignment tasks between the fairly cumbersome SpinView and some more specialized acquisition software we wrote that runs off the Julia Spinnaker wrapper.  We had to update the Spinnaker SDK to the latest (as of a few weeks ago) 2.20.48 to get around some driver crashing issues, but now we can't use the micro-manager interface.

Thanks,
Rodrigo

nicost, 2020-12-22:
See: https://forum.image.sc/t/fast-time-series-acquisition-for-10gige-camera/44418/20

Basic problem is that Micro-Manager Spinnaker support may "break" whenever FLIR updates their SDK, and we are legally not allowed to redistribute a specific version of the SDK.  CAIRN was allowed to re-distribute a specific version in use at the time, so that is why we are frozen at that version.  I'll contact FLIR...

ischoegl, 2021-02-05:
Hi @nicost and @rguerrab ... I was wondering whether there’s any update for spinnaker support in the works at the moment? We just received a FLIR Oryx that we’d like to use with micromanager eventually. It’ll probably just work with an older SDK for the time being, but I wanted to ask before setting up our tool chains.

nicost, 2021-02-05:
Yes, this is on my todo list.

nicost, 2021-03-08:
MM is build using the V2 Spinnaker SDK as of 2021-03-08. The downside may be that the camera no longer works for people who do not upgrade the Spinnaker driver.  I was told that it should work with every 2.0 driver (as long as they support vs2010).

If there are any issues, please open a new ticket.

closed 2021-03-08 by nicost

navidroo, 2021-06-04:
Hello,

I want to configure micro-manger to be connected to Flir-Oryx-10GS-32S4C-C. My micromanger version is 2.0.0-gamma1 20210602 and spinnaker SDK version is SpinnakerSDK_Full_2.40.144_x64. I installed the packages for both Visual studio 2010 and 2015 in spinnaker. What I did (in micromanager) was to go to Devices-->Hardware Configuration Wizard-->Create new config--> Find spinnaker from the list-->Select the camera and click on adds. All the information in the new window is consistent with camera information (SN, Model). But when I click on OK an error happens:

“Error in device “Oryx-10GS-32S4C-C”: Spinnaker: GenICam::Logical ErrorException=Unable to execute because ‘AcquisitionStart’ is not writeable : LogicalErrorException thrown in node ‘AcquisitionStart’ while calling ‘AcquisitionStart.Execute()’ (file ‘Command.cpp’, line 109) [-2005] (10002)”

I was been able to acquire images using Spinview software. 
Could you please tell me how to fix the error?

JosephGreene, 2021-06-12:
Hi @nicost,

I was hoping to follow up on the comment left by @navidroo. I am using a FLIR-BFS-U3-04S2C-CS camera (Blackfly S) with the latest micromanager 2.0.0-gamma x64 build and SpinnakerSDK_Full_2.40.144_x64. However I am receiving the exact same issue indicating that 'AcquisitionStart' is not writable. 

Currently, my intention is to automate a procedure with pycromanager, which requires micromanager gamma, so I can't backtrack to beta. In addition, I tried Spinnaker 1.2.0 and Spinnaker 2.0.0, which failed to recognize the camera and crashed the software respectively. Any help you can offer will be sincerely appreciated.

nicost, 2021-06-14:
We are building the device adapter against version 2.30.  Could you download and install that version (it is in the archive of FLIR Spinnaker download) and see if that works?  

Also, can you please open a new ticket, this closed one is very hard to find.

JosephGreene, 2021-06-15:
@nicost Yes! I will open one now. I checked the available Spinnaker SDKs from the FLIR Archive and it looks like 2.3.0 is unavailable for the full SDK. I did reach out to the company to see if they could provide it

navidroo, 2021-06-15:
Hi @JosephGreene 

I tried different versions of Spinnaked and mm and finally it worked. The versions I used are:
mm 2.0.0 gamma1 20210320
spinnaker SDK 2.3.0.77 (x64)
Also select VS 2010 and 2015 when installing SDK.

JosephGreene, 2021-06-15:
Hi @navidroo,

That's great! Let me give it a shot. I also noticed I misspoke and 2.3.0.77 is available on the achieve, I missed it the first time through. Thanks for posting your solution!