micro-manager/micro-manager #1068: How to propagate hardware feedback?

wwarriner, 2020-12-28:
A lab I work with has a Jenoptik Gryphax Prokyon camera and I am writing an adapter for it that is compatible with the recent version of MicroManager.

There are two specific issues where the hardware changes properties that I believe should be fed back up to the MicroManager GUI, but I don't see any way to achieve that.

First, binning may not be set directly for this camera. Instead, it is controlled indirectly by changing the hardware imaging mode. Ok, fine, not an issue, I have hooked up the properties so they play nicely. My issue centers on the fact that I am still able to modify the binning value in the GUI. I have the SetBinning() function return DEVICE_CAN_NOT_SET_PROPERTY, so a modal error dialog box appears when it is modified. However, the original value is not restored. How can I automatically restore the value when it is changed? I want to at least simulate a non-editable combo box, or make the binning combo box not editable.

Second, the camera hardware has limitations on ROI placement that are a few pixels smaller than the nominal resolution in some imaging modes. Internally I clip the ROI to the hardware limits, but from what I can see there is no way to propagate the values back up to the MM GUI. This is important because it is also possible for the hardware to adjust the ROI values if, say, the imaging mode is changed, which also changes the resolution. This could result in the GUI ROI becoming invalidated.

Concretely, suppose the resolution is 1920x1080. The image index limits would be (0, 0, 1919, 1079), while the hardware ROI limits might be (2, 2, 1917, 1077). If the user selects an ROI of (0, 0, 1919, 1079) then the values are clipped internally to the hardware limits. However the GUI ROI shows what the user selected, rather than the actual internal state of the adapter and hardware.

I appreciate your thoughts.

Thanks for all your hard work and support!

nicost, 2021-01-05:
Sorry for the delay in getting back to you!  Do the ROI changes only take place after the user sets a new ROI?  In that case, the UI could solve this by asking the hardware for the current ROI after setting the ROI.  That seems a prudent thing to do in any case (and I have vague memories that approach was used in the past, but clearly, the code does not do so currently).  

If ROI changes can also occur at other moment, then a callback would be in order.  I do not think that a callback signalling the current ROI is currently available, so if it is needed, it will have to be written.

nicost, 2021-01-27:
OK to close or would you like more discussion on this?

wwarriner, 2021-01-27:
I would love more discussion on this but I don't think time is on my side right now, and this is fairly low on my priority list. Would it be acceptable to reopen it at a later time? Or prefer a new issue?

nicost, 2021-01-27:
Sure, re-opening is fine (just would like to keep issues actionable for us;)

closed 2021-01-27 by nicost