micro-manager/micro-manager #1077: Inefficient refreshing of property cache due to OnPropertiesChanged event.

nanthony21, 2021-01-09:
While the main purpose of caching property values is to improve performance, if the cache is refreshed too frequently it can do more harm than good. On systems with many connected devices refreshing the cache can result in noticeable freezing of the GUI.

The `onPropertiesChanged` handler of `CoreEventCallback.java` involves a full update of the property cache.  This can be particularly problematic with certain device adapters which make frequent calls to `OnPropertiesChanged()`. The PVCam adapter is one that comes to mind. Even just changing the exposure seems to cause the whole cache to be refreshed.

Even though the java `CoreEventCallback::onPropertiesChanged` method refreshes the whole cache it is worth noting that the C++ function that triggers the java event callback, `CoreCallback::OnPropertiesChanged(MM::Device* caller);` contains the following comment: 
```
   // TODO It is inconsistent that we do not update the system state cache in
   // this case. However, doing so would be time-consuming (if not unsafe).
```

I propose that `CoreCallback::OnPropertiesChanged` should have proper cache updating added to it's implementation such that only the properties of the `MM::Device` that called it are updated. Then `CoreEventCallback.java` will no longer need to go through a full update of the property cache. Is there a reason that this change should not be made?

nicost, 2021-01-11:
I do believe that more fine-grained notifications are in order.  However, re-interpreting the meaning or "OnPropertiesChanged" does not seem right.  That is a blanket call that implies that anything may have changed.  For instance, it could be issued by a Hub device, and only querying that hub device will miss changes in properties in its associated child devices.  So, I would rather make a new function that signals that all properties of the device should be queried. 

As for the PVCAM adapter, it may well be possible that it is known which properties could potentially change, and the blanket call to OnPropertiesChanged could be replaced with a series of specific OnPropertyChanged function calls.  For instance, in your example of changing the exposure time, what could the reason be to call OnPropertiesChanged?

nanthony21, 2021-01-11:
@nicost, thankyou for the feedback.

I would argue that this wouldn't mean re-interpreting the meaning of the `OnPropertiesChanged` event but would actually just mean completing the implementation of `CoreCallback::OnPropertiesChanged(MM::Device* caller)` as it was originally intended (as far as I can tell from the function signature and comments).  This would ultimately have the added benefit that `MMStudio` would no longer have a need to handle the `OnPropertiesChanged` event in such an inefficient manner.

The case that you mention about a Hub device is a good edge-case which I think can be handled entirely within the C++ layer in `CoreCallback::OnPropertiesChanged(MM::Device* caller);`. As far as I'm aware there are no other special situations like this that would need to be handled.

Here are links to the relevant code:
https://github.com/micro-manager/micro-manager/blob/961eea3e5f8e95cad380dcab97e04874c595ffc1/MMCore/CoreCallback.cpp#L449

https://github.com/micro-manager/micro-manager/blob/961eea3e5f8e95cad380dcab97e04874c595ffc1/mmstudio/src/main/java/org/micromanager/events/internal/CoreEventCallback.java#L72

I don't have access to a PVCAM device right now to do further testing so I am working completely from log files. It appears that there are many circumstances that can cause `PVCAMUniversal.cpp` to call `coreCallback->onPropertiesChanged(this);`. The most likely culprit is the `Universal::applyAcqConfig` method which appears to be called from 38 other methods in the device. I imagine there are probably ways to improve upon the efficiency of the code in this ~6000 line file but I think that it will be much more complicated and ultimately less beneficial than the changes proposed above.

nanthony21, 2021-01-12:
Just to be clear, I'm not asking for someone else to do this. I'm planning to work on this myself and and submit a pull request, I just wanted to make sure that such a change wouldn't be rejected outright. @marktsuchida , do you have an opinion on this?

nicost, 2021-01-12:
Sounds good!  Please note that for the time being all changes to the C++ code need to be made in the subversion repository.  It may be easiest if you do this in a separate branch, and I can port to the subversion repo.  Mixing Java and C++ changes in one branch will make it very difficult for me to merge things though.

marktsuchida, 2021-03-15:
Sorry for the belated comment. I think a new and clean and backward-compatible (if at all possible) mechanism is clearly needed, because people have made various assumptions about `OnPropertiesChanged` (such as that it doesn't take a super long time to call). I've encountered problems with this function many times, and (although it would be easy to come up with a better implementation if starting from scratch) I have to say I have not so far been able to think of a solution that would not break lots of existing code -- at least not one that would be simple enough to be practical. Part of the problem is that the behavior of the system state cache itself is poorly defined (or, to put it more kindly, it's a "best effort" feature in terms of accuracy), yet lots of Java code depends on the current behavior for reasonable performance (if not accuracy under specific assumptions).

nanthony21, 2021-03-15:
Thanks Mark,

Do you think it would be a problem to do what is proposed above?: Make it so that a call to `OnPropertiesChanged` refreshes the properties of only the calling device, or maybe of all devices that share the same `HubDevice`.

marktsuchida, 2021-03-16:
@nanthony21 I don't in principle see a problem with limiting the cache update to the same (or closely related) device(s) -- that sounds like a good idea. But I'm a little worried about the cases in `CoreEventCallback.onPropertiesChanged()` where the cache update is skipped. If the cache update is moved into `CoreCallback`, it will not be possible to skip the update under those same conditions. Whether that is a true problem is hard to say, because it probably depends on the device, and we have about a dozen devices that call `OnPropertiesChanged()`, most of which I have never worked with. I think I would feel somewhat safe if it doesn't cause issues with PVCAM under all of the relevant conditions (MDA and loading config?), and the code for all the other devices is reviewed.

nicost, 2021-05-13:
Hi @nanthony21, did you start working on this?  I am thinking a bit about tackling this (even before the 2.0 release), as it will offer significant speed up for some systems (and I'll get some PVCAM cameras myself soon;).

nanthony21, 2021-05-13:
Hi @nicost . I started on it very briefly but never got far enough to have anything worth saving. So if you're able to take this on, that would be wonderful!

nicost, 2021-05-18:
@marktsuchida, can you elaborate about "But I'm a little worried about the cases in CoreEventCallback.onPropertiesChanged() where the cache update is skipped"?  What are those cases?  

It looks like the Core callback "onPropertiesChanged" results in a call to CoreEventCallback.onPropertiesChanged() on the Java side, which results a full cache update.  I therefore do not understand how skipping a cache update can happen (in the Java environment).

marktsuchida, 2021-05-18:
I think there is a chance that it won't end up being a problem, but I was referring to the `if` statement in `CoreEventCallback.onPropertiesChanged()`, lest @nanthony21's earlier summary leave the impression that it unconditionally updates the state cache. It only updates the system state cache and sends downstream events when an acquisition is not running _and_ it is not disabled via `setIgnoring()`, which sets `ignorePropertyChanges_`.
https://github.com/micro-manager/micro-manager/blob/961eea3e5f8e95cad380dcab97e04874c595ffc1/mmstudio/src/main/java/org/micromanager/events/internal/CoreEventCallback.java#L63-L77

Searching for usage of `setIgnoring()` shows it is only used in one place: `MMStudio.loadSystemConfiguration()` arranges to skip the update to system state cache and the downstream event while loading the hardware config. So maybe this can be dealt with in the Core (if it is indeed necessary).

For the acquisition running case, I'm not sure what the reason was (though I imagine it was speed), and I'm not sure if the true intent was to suppress the event notifications or the cache update or both. Hopefully the state cache can be updated even in this case if it is made faster by limiting to the same device, but it probably needs to be tested carefully under MDA (but I guess that was the case anyway).

tomhanak, 2021-06-04:
Just found this issue...

Unfortunately, in PVCAM adapter code we need to call the ```OnPropertiesChanged``` because this seems to be the only way that correctly updates the property range in the UI. The ```OnPropertyChanged``` updates the value only. In most cases we know what properties could change the value, and if there would be a way how to refresh values and/or ranges of selected properties only, we would use use it instead.

Simply take a look at ```OnPropertiesChanged``` calls in PVCAM adapter, every occurrence has attached a comment saying why it is needed. Sometimes we cannot call it (the call it in code but commented out) because it would break or hang the app, so we rather keep some properties out-dated.

nanthony21, 2021-06-04:
> we need to call the OnPropertiesChanged because this seems to be the only way that correctly updates the property range in the UI

I didn't realize that, this is good to know.

I think the PVCam device adapter is what caused me to notice this issue in the first place. The frequent `OnPropertiesChanged` calls of a PVCam device coupled with a few slow 9600 baud devices can result in very laggy performance.