micro-manager/micro-manager #1090: API does not allow updating positions in the current positionList

nicost, 2021-01-28:
When executing a multi-position acquisition, it may be beneficial to update the positions in the position list, for instance to correct for focus drift. The current API has not methods (that I can find) to do so.  The current acquisition engine takes the acquisition list when its starts an acquisition, so calls to the PositionListManager (which hands out a copy of the positionlist rather than the positionlist itself, and which can set a new one, which will be picked up by the PositionListDialog, but not by the already running MDA.

I am not sure what the best solution is.  Clearly, giving direct access to the PositionList is not desirable, but it will also be difficult to convince (and likely undesirable) to convince the acquisition engine that there is a new position list (for instance, what should it do when the names of the positions, or the number of positions changes?).  One solution may be to have the API provide a mostly immutable version of the current PositionList, that does allow to change the actual stage positions, but nothing else.

marktsuchida, 2021-04-28:
Is it that you want to modify the current position list while an acquisition is running, but only for the purpose of using it in a subsequent acquisition? Or do you want to actually modify the positions used by an ongoing acquisition?

The latter would require new development in `acqEngine`; I'm not sure we'd want to embark on such a project.

nicost, 2021-04-30:
Are you sure that new development in the acqEngine is needed?  In the past we were updating the position list (for instance, we were updating the z position to move to after going to a position, locking in the PFS and reading back the Z position so that we would go to a better position the next time).  The goal here is exactly the same, update a Z position based on better, now accurate information.  I do believe the the acquisition engine works fine with that, but that the immutability of the Positions in the list is in the way.

marktsuchida, 2021-05-05:
The behavior you describe is correct, except that the update does not happen in MMStudio's `PositionList` but rather only in acqEngine's internal state. (Leading to the unrelated problem of users never realizing that this useful behavior exists.)

Every time after an autofocus (which might be PFS) is run, it stores the arrived Z position under the `:last-stage-positions` key of its `state`:
https://github.com/micro-manager/micro-manager/blob/982035adab055eede5c5979975bded83a8b86417/acqEngine/src/main/clj/org/micromanager/acq_engine.clj#L270-L271