micro-manager/micro-manager #1094: Invalid OME-XML produced by MicroManager

JacksonMaxfield, 2021-01-30:
Hello all!

I maintain a project called [aicsimageio](https://github.com/AllenCellModeling/aicsimageio) which for the context of this issue simply is: "a pure python library for microscopy image file format reading."

We have an `OmeTiffReader` and in the process of upgrading to our new 4.0 API we made the decision to adopt a library that specifically validates that any XML provided in is _valid_ OME XML (to the XSD).

We had an issue posted earlier today that tested a micromanager produced OME-TIFF and saw that it resulted in an `XMLSchemaValidationError`. Full issue here: https://github.com/AllenCellModeling/aicsimageio/issues/195

The shorthand, is that it looks like the `InstrumentID` in OME-XML produced by micromanager isn't valid to the OME XSD.
I tracked down what I think is the [single line culprit](https://github.com/micro-manager/micro-manager/blob/bc3f1d42ca3f18d721b7c268064436600bb3ea8c/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/OMEMetadata.java#L175).

I _think_ this should be:
```java
String instrumentID = MetadataTools.createLSID("Instrument", 0);
``` 

as specified in the [bioformats API](https://downloads.openmicroscopy.org/bio-formats/5.0.6/api/loci/formats/MetadataTools.html#createLSID(java.lang.String,%20int...)).

Sidenote, there is only one other reference to `createLSID` and it is used for the [camera](https://github.com/micro-manager/micro-manager/blob/bc3f1d42ca3f18d721b7c268064436600bb3ea8c/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/OMEMetadata.java#L394) I am unsure if that may need to be updated too.

It would be great to support micromanager OME-TIFFs in `aicsimageio` so hoping this spec compliance can be resolved :slightly_smiling_face:

nicost, 2021-01-30:
Thanks for looking into this @JacksonMaxfield !  I can't say that I understand the OME documentation very much (what is an LSID, what is an Instrument, and why should it have an ID with one or more indices, but the API allows for zero indices, which then generates and invalid data structure? Weird...), but made some changes as you suggested.  Can you have a quick look, and if you think this looks OK, I can push it into the main branch and it will appear in the next nightly build.  Completely agree that it will be nice if this is correct, and if there are more suggestions to make the OME-XML that the MM code generates more useful, I'll be happy to incorporate them.

JacksonMaxfield, 2021-01-30:
> why should it have an ID with one or more indices

Heh. I also am not the best person to ask but I have seen it used as a way of representing scene / sub-Image management.

I.E. "Instrument:0:0" is the instrument used for the 0th scene and "Instrument:1:0" is the instrument used for the 1st scene, etc.

JacksonMaxfield, 2021-01-30:
I just manually edited the file metadata given to me as a part of the issue report and can confirm, changing the "Microscope" to "Instrument:0" and the "Camera" to "Detector:0" resolved the issues. :tada:

JacksonMaxfield, 2021-02-01:
Closing as now resolved! Thanks @nicost

closed 2021-02-01 by JacksonMaxfield

tlambert03, 2021-02-02:
just catching up on this issue.  @nicost, the OME library @JacksonMaxfield is using is https://github.com/tlambert03/ome-types

> if there are more suggestions to make the OME-XML that the MM code generates more useful, I'll be happy to incorporate them.

If you happen to have a big-ol' store of example XML output from MM, (or if they can be autogenerated?) I could probably assist finding any residual issues :joy: ?

as for those pesky LSID issues, all OME "`*IDs`" derive from the [LSID type](https://www.openmicroscopy.org/Schemas/Documentation/Generated/OME-2016-06/ome_xsd.html#LSID), and the takehome here is that they each specify a `facets.pattern` regex that more or less requires them each to start with the name of the type.  for instance, the [InstrumentID](https://www.openmicroscopy.org/Schemas/Documentation/Generated/OME-2016-06/ome_xsd.html#InstrumentID) regex pattern is `((urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:\S+:\S+)|(\S+:\S+)) & ((urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:Instrument:\S+)|(Instrument:\S+))` 🤢