micro-manager/micro-manager #1104: Discussion: Build System: Migrating and cleaning the `3rdPartyPublic`

nanthony21, 2021-02-12:
It would ultimately be nice to move the Micro-Manager build process onto a CI/CD service like "Github Actions". However, before this can happen a few things need to happen to simplify setting up the build environment. One of these things that need to happen is to make it easier to get the dependencies that are currently stored in the `3rdPartyPublic` SVN repository. The repository is currently 6.68GB, some of that is SVN history but 3.6GB are from files still present in the repository.

I'm hoping to get some opinions on the best path forward.

### Where to store the dependencies.
There are many different ways that these files could be stored and it may be that SVN is still the best option. Since the source code is primarily managed on Github I would suggest that it would make sense to also store these dependencies in a Git repository. The dependencies repository could be added as a submodule to the Micro-Manager repository. That way all that would need to be done to fetch the dependencies after cloning the Micro-Manager repository would be to run `git submodule init` and `git submodule update`. This would also allow for linking a specific commit of the the dependencies repository to a specific commit of Micro-Manager. However, this would first require that we greatly reduce the size of the repository.

### Cleaning the repository
Tracing which contents of the repository are actually needed can be a little difficult but I think there must be files that are not being used. For example, we have Boost 1.55 with binaries for  various platforms and we also have the source for Boost 1.46.1. There are also two versions of OpenCV. I would suggest that whenever possible we should fetch these dependencies from online  (like is done for the MMStudio Java dependencies) rather than managing them in our own repository but in some cases that won't be possible.

Here is a list of the current contents. Any feedback on files that aren't needed are appreciated.

- [x] bb64_complete.bat
- [x] bbminimal32.bat
- [x] bbminimal64-vs2010.bat
- [x] bbminimal64.bat
- [x] boost
- [x] boost-versions
- [x] classext
- [x] clojure_interpreter
- [x] dc1394-drivers
- [x] FreeImage
- [x] google
- [x] hidapi
- [x] JavaLauncher
- [x] javalib3d
- [x] Kurios
- [x] libdc1394
- [x] libusb-win32-bin-1.2.2.0
- [x] libusb-win32-bin-1.2.3.0
- [x] libusb-win32-bin-1.2.6.0
- [x] loci
- [x] modbus
- [x] Motic
- [x] OpenCV2.3
- [x] OpenCV2.4.8
- [x] Photometrics
- [x] Picard
- [x] pugixml
- [x] sourceext
- [x] steinbach
- [x] swig
- [x] tags
- [x] TUCam
- [x] zlib

nanthony21, 2021-02-12:
I forgot that I had done this, but I actually started on this project a long time ago. Here is the resulting Git repository: https://github.com/nanthony21/MM3rdparty

There are still a few things that will need to be done before it can be used though.

nicost, 2021-02-17:
That looks like a great step forward! We could move it to micromanager/3rdpartypublic (or another name).  Since there also is the "3rdparty" repo in svn (for material that we are not allowed to make publicly available), we should stick to a name that distinguishes the two.

marktsuchida, 2021-02-18:
> Any feedback on files that aren't needed are appreciated.

This can only be reliably determined by deleting things and checking that the results of the build do not change from the current Windows and Mac nightly builds. (Merely checking that the build appears to succeed is not sufficient, of course.)

If setting up a new way to build, I would suggest considering using an umbrella build project that has `micro-manager` as one of its submodules, rather than a "3rdpartypublic" repo. The idea is that the generic (configurable) build system (autoconf/automake, VisualStudio, and Ant) belongs to the main project, while the task of "building a full-featured binary distribution" is a significant undertaking in itself, worthy of its own Git repo. In fact, there should be separate "build" projects for Windows and Mac, because the handling of dependencies is completely different in most cases.

New versions or alternatives of the scripts that are currently in `buildscripts/nightly` could be added to the "build" project to set up a new CI, while keeping the current CI unaffected for as long as it is needed.

Another advantage is that third-party dependencies can be pulled in by whatever mechanism that is suitable: properly structured dependencies like boost and zlib can be downloaded as source and built, rather than being vendored(*) copies or submodules, making upgrades much simpler. This is similar to how the `pymmcore` project works.

(*) - see, e.g., https://gist.github.com/datagrok/8577287

nanthony21, 2021-02-18:
Thanks @marktsuchida . 

I have a branch on my fork called "3rdPartySubmodule" that implements this. And has removed all of the unused files. It also includes boost as a submodule and builds from source. This greatly increases the build time but I think is still better than storing the binaries.

However Boost doesn't seem to support VS2010 so this branch won't really work until the build process allows for use of a newer version of Visual Studio.

On a related note do you think it would make sense to split the C++ code into it's own Git repository that is included as a submodule in the Java and Python repositories? That way the C++ repo could have it's own build scripts that aren't tangled up with all of the Java build scripts.

marktsuchida, 2022-01-21:
We have agreed to keep `3rdpartypublic` as an SVN repo for as long as it is needed by the current build systems and the current (Jenkins) CI. When we transition to individual Git repositories for each C++ project (with Meson build scripts and GitHub Actions CI), dependencies will be moved to their own individual Git repositories where necessary (or be built as part of the project's CI in the case of open source dependencies).

Thus, for example, the prospective `mmdev-HIDManager` repo's CI should build its own copy of the `hidapi` static library. Building open source dependencies like this can be done by Meson as subprojects (probably via wrap files, so as to avoid Git submodules as much as possible), but the default should be to use system-installed libraries (at least on Mac/Linux). Hopefully the same principle will work for Boost, although special handling may be needed.

For closed source dependencies, the CI will use the binaries in (mostly private) repos, again probably via Meson wrap files. For these, too, the Meson build scripts should default to finding dependencies installed on the local system, so that people can do what is most natural: install a vendor SDK in its default location, clone a single device adapter, and built it, without having to edit paths. In those few cases where the vendor does not provide an installer for there SDK and therefore there is no clear install location, it will be necessary to point Meson to the installation using command line options.

In other words, `3rdpartypublic` (and `3rdparty`) will be eliminated. This has several advantages:
- No need to clone or check out a large repository of which only a small fraction is needed for many tasks (such as modifying a single device adapter)
- No need to (figure out how to) edit path settings in order to try out building with a different version of a dependency from the one officially provided
- Ability to build device adapters that require vendor SDKs that we cannot redistribute, without (usually) having to edit path settings
- No possibility of being affected by dependency issues of other modules when trying to build a particular device adapter (as in the current Linux/Mac build)

closed 2022-01-21 by marktsuchida