micro-manager/micro-manager #1106: Micro-Magellan plugin Export Grid to Micro-Manager does not work

nicost, 2021-02-13:
In the Micro-Magellan explorer, create a grid (or multiples), click "Export selected to Micro-Manager", or "Export all", and nothing appears in Micro-Manager.  I would expect these to show up in the position list.  The position list code was cleaned up recently, so it is possible that the code went out of sync.  No exceptions in the corelog though. One of our lab members is planning on using this feature soon.

henrypinkard, 2021-02-13:
I haven't changed the code. Must have been something that changed on the micro-manager API side. The relevant code is here:

https://github.com/micro-manager/micro-manager/blob/9934be60b66ca4fa994cbcd61f7a12c4b5c3b6b8/plugins/Magellan/src/main/java/org/micromanager/magellan/internal/surfacesandregions/XYFootprint.java#L49

nicost, 2021-02-14:
Yes, the api (https://valelab4.ucsf.edu/~MM/doc-2.0.0-gamma/mmstudio/org/micromanager/PositionListManager.html) is now implemented as intended, and the call:
`Magellan.getStudio().positions().getPositionList();`
returns a copy of the poitionlist.

I think that changing to:
```
PositionList posList = Magellan.getStudio().positions().getPositionList();
posList.add(mPos);
Magellan.getStudio().positions().setPositionList(posList);
```
will do the trick.

henrypinkard, 2021-02-14:
https://github.com/micro-manager/micro-manager/pull/1108

nicost, 2021-02-16:
Should be fixed now in PR #1108

closed 2021-02-16 by nicost