micro-manager/micro-manager #1107: NIDAQ having unknown error

joaomamede, 2021-02-14:
I can't configure my NIDAQ , either analogIO or Digital IO
I have a PCIe-6323 and all the drivers installed and the system is working through NIS-elements.

I tried to manually set the parameters, even adding the Demo to Yes  in AnalogIO. I always have the error

`Error in device "AnalogIO":(Error message unavailable) (-200220)`

What am I missing?
[image]

[image]

Thanks!

nicost, 2021-02-14:
You can try using the NIMultiAnalog device adapter instead.  It is newer.

joaomamede, 2021-02-17:
Thank you. I believe it worked (it detected ao0 through 4). I now have to learn how to configure it with the NIDAQ and NIBB in my TIe2. Thanks.

nicost, 2021-02-17:
Great!  Curious to hear how it goes.

joaomamede, 2021-02-17:
Any hints where to start (I'd be very gratefull just by a linking to a right html)? 
I read all the docs in the website, but for example, how do I assign the right NIDAQ pins to colors and my emission filters?

[image]
[image]

nicost, 2021-02-17:
I have never worked with these myself, so it is difficult to give you exact steps.  First look in the property list of the analog out device if there are ways to set the output voltages of the output pins directly.  If there are, use these to see if you can use them to control the SpectraX and the filterwheel.  Next, go back to the Hardware Configuration Wizard, and add the DATTLStateDevice from the Utilities device adapter.  The DATTLStateDevice can have up to 8 output pins (no idea if you can assign those randomly as in your list, or that they need to be consecutive), so that a given state in the DATTLStateDevice corresponds with certain pins being high and others being low (for instance, the number "5" will set pins 0 and 2 high and leave the other 6 low).  You probably need to make two of those, one for the SpectraX and the other for the Filter Wheel.  Finally, you need to add one StateDeviceShutters, that attaches to the SpectraX DATTLStateDevices and become a virtual shutters for the SpectraX.

Complicated!  If you get it to work, it would be great if you could write a tutorial.

joaomamede, 2021-04-11:
I finally had the scope not imaging to try this again.

[image]

I added the NIMultianalog and it seems to pick it up.
I could not find ways to setup individual pins through the interface before adding the DATTLStateDevice.

I added the DATTLStatedevice, I think in a previous test I could set the pins from 0 to 255, but that option is not here anymore.
[image]

Am I very off-track? Thanks for the help. If I get this working I'll def make a tutorial and video.

nicost, 2021-07-22:
You will need to set the number of DA devices to be used by the DA TTL State device in the Hardware Configuration Wizard.  If you set it to 8, the number of possible states will be 255.  You will also need to assign each of the individual DA devices (best done in the System-Startup group).  

Also note that I just pushed an update to the DA TTL State Device.  I expect it to be much easier to work with now.  

It certainly is way too complicated to set this up.  I'll think a bit how to simplify this work flow.

ptbrown1729, 2021-08-03:
It seems like you have arrived an alternative solution, but thought I would chime in anyway. 

I am using a similar NI card (PCIe-6343) with the NI100x device adapter. Based on my experience with that, I think you need to set the "IOChannel" property. In your screen shots it looks like this is still "undef". For the digital line this should be "Dev1/port0"

joaomamede, 2021-08-20:
> It seems like you have arrived an alternative solution, but thought I would chime in anyway.
>
> I am using a similar NI card (PCIe-6343) with the NI100x device adapter. Based on my experience with [...]

Indeed. I had managed to turn the LEDs on doing that in the last few months.
I still get very confused with the 256bit translation code @nicost  helped me in a image.sc post. But the scope has been 99% busy with imaging and I couldn't try others.

I need a translation table :)

closed 2021-12-01 by nicost