micro-manager/micro-manager #1109: Clean-up / Simplify project folder structure.

nanthony21, 2021-02-16:
I realized that even though I am fairly familiar with the Micro-Manager source code, there are still many folders/files in the repository that I don't know the purpose of. Many of these folders contain only one or two files and I can't tell if they are actually used for anything. I would like to propose that some related folders be combined together and also wanted to see if there are any that can be gotten rid of. Can anyone more familiar with the code explain the purpose of the folders at the end of my list?

## Folders that make sense to me
- [ ] acqEngine: The acquisition engine behind the MDA. Written in Clojure.
- [ ] autofocus: Java AutoFocusPlugins accessible from MMStudio.
- [ ] bindist: Binary files which are copied to the `stage` directory during the build process
- [ ] buildscripts: Ant Scripts and other files used during the build process.
- [ ] DeviceAdapters: C++ device adapters providing compatibility with various hardware.
- [ ] libraries: Java libraries that are used by other components of the project.
- [ ] MMCore: The C++ Core which facilitates all hardware control.
- [ ] MMCoreJ_wrap: Java bindings for MMCore
- [ ] MMCorePy_wrap: Python bindings for MMCore
- [ ] MMDevice: The MMDevice interface which defines how MMCore interacts with hardware devices.
- [ ] mmstudio: A Java GUI which adds additional functionality on top of MMCore.
- [ ] plugins: A collection of Java plugins for the `mmstudio` project.

##  Folders that I think should be combined
*These are both essentially tutorial folders. One contains MMCoreJ examples and one contains MMStudio examples.*
- [ ] scripts: Useful example scripts for beanshell.
- [ ] Tutorial: A collection of Java examples showing how to use MMCoreJ.

*These folders both contain documentation*
- [ ] doc: Documentation source files.
- [ ] doxygen: Documentation for MMCore

*These are all test folders*
- [ ] systemtest
- [ ] TestDeviceAdapters
- [ ] testing

## Folders that I don't know the purpose of
- [ ] swig-doc-converter
- [ ] server
- [ ] mmAsImageJMacros
- [ ] ipch
- [ ] Util: A single c++ file which may be used elsewhere in the project.
- [ ] portdebian
- [ ] MacInstaller
- [ ] m4
- [ ] drivers: Driver installers for some Nikon equipment.

nicost, 2021-02-16:
>*These are both essentially tutorial folders. One contains MMCoreJ examples and one contains MMStudio examples.*
>- [ ] scripts: Useful example scripts for beanshell.
>- [ ] Tutorial: A collection of Java examples showing how to use MMCoreJ.

Tutorials is very old, has not been maintained, and should probably be removed.  Scripts is a collection of (useful?) starting points for people to write their own scripts.  They are distributed with MM, and I personally regularly use them to remind me how to start a quick utility beanshell script.

>- [ ] doc: Documentation source files.
Looks like an old copyright notice and build instructions that are also on the website.  Those build instructions could be useful, but it would be good to have a "single source" of them rather than both on the website and in the source code.
>- [ ] doxygen: Documentation for MMCore
Placeholder used when generating doxygen.  I guess these could be generated when needed, but that needs some changes to the build.

>*These are all test folders*
Different attempts from different eras to include automated tests.  All good efforts but none of them ever rose to the level of inclusion in the build process.  I guess these could be archived.

>- [ ] swig-doc-converter
I think this is the code that generates this (https://valelab4.ucsf.edu/~MM/doc-2.0.0-gamma/mmcorej/mmcorej/CMMCore.html) documentation.  It runs on the Mac build machine.  Still useful.
> - [ ] server
php scripts used on the web server to store and process uploaded config files and trouble reports (the help > report a problem uploads a corelog and send emails to several micro-manager email addresses).
>- [ ] mmAsImageJMacros
Code to make Micro-Manager recordable by the ImageJ macro recorder (I forgot who the author is). It is included in the MM distribution for those who like this functionality.
> - [ ] ipch
This directory seems to be used whenever I build (debug mode?) code on my laptop.  Some kind of intermediate files from VS2010?  Possibly files with symbols?
 >- [ ] Util: A single c++ file which may be used elsewhere in the project.
Brent.h.  We may want to grep the source code to find where it is used, and decide if there is a better place.
>- [ ] portdebian
I have no idea if anyone still uses this.
>- [ ] MacInstaller
Used to generate the Mac installer.  Could live elsewhere, but then would need to change the Mac build process
- [ ] m4
m4 macros used by the autotools used for the build on unix (Mac/Linux)
>- [ ] drivers: Driver installers for some Nikon equipment.
libusb drivers for two Velleman boards and two Nikon microscopes.  Agree these should not be in the source code.  It would be better to distribute these through the micro-manager website.

nanthony21, 2021-02-17:
These Pull Requests address this issue:
#1116 
#1113
#1112
#1110

closed 2021-02-17 by nanthony21