micro-manager/micro-manager #1115: Move the drivers folder in the source code repo to a better place

nicost, 2021-02-17:
The "drivers" folder contains libusb-based drivers and code to re-generates them for the Velleman K8055, K8061 and Nikon AZ100 and Eclipse90i.  These files should not be part of the source code repository, but they should remain available somehow, so we need a good home for them.

marktsuchida, 2021-02-18:
Just a new repo under github.com/micro-manager?

nicost, 2021-02-19:
Great idea!  Done in #1120 (and by creating the new repo).  CCing @nanthony21 so that he is aware.  Hope that the several drivers folders removals and re-additions do not lead to trouble for folks.  I had to reboot my computer because of permission issues with that folder.  Suspect it uncovered a small bug in git/bash for Windows.

closed 2021-02-19 by nicost