micro-manager/micro-manager #1118: Device Adapter Bloat: Duplicate X-Cite device adapters.

nanthony21, 2021-02-18:
Choosing the correct XCite device adapter to use is currently difficult, there are 4 separate libraries which seem to have overlap in which devices they control.

The `XCiteXT600` adapter appears to be a direct copy-paste of `XCiteLED` with just a few lines added. It should be determined if the "XT600" adapter still applies to the same devices as the "XCiteLED" adapter and if so the old version should be removed.

nanthony21, 2021-02-18:
It seems like Tom Porter at Excelitas wrote the XT600 adapter and is probably the best person to contact about this. I'm not sure how to get a hold of him though.

nicost, 2021-02-26:
The issue with removing Device Adapters is that people's configurations will stop working.  There may be a way to make them "invisible" in the Hardware Configuration Wizard.  I contacted Tom.

Also, it seems that the XCite120PC_Exacte supersedes the Xcite adapter.

nanthony21, 2021-03-01:
That makes sense, this definitely isn't a huge priority. Eventually it will become to difficult to maintain all the different projects though. The serial commands used by all of the various XCite devices are similar enough that it would make sense to have them all combined into a single project with only 1 or 2 different device adapters.

closed 2021-03-01 by nanthony21