micro-manager/micro-manager #1122: MMDevice Shared vs Static Runtime

nanthony21, 2021-02-22:
There are two projects for  MMDevice, "SharedRuntime" and "StaticRuntime". However, the Visual Studio projects for both of these are configured to compile as static .lib files.

Is it meant to be this way or this a mistake?

The only difference I see between the two projects is that the "StaticRuntime" has the following property defined: `<RuntimeLibrary>MultiThreadedDebug</RuntimeLibrary>`

nicost, 2021-02-23:
Great questions!  @marktsuchida do you recall what these are and what they were/are for?

marktsuchida, 2021-03-15:
Not a mistake, it is essential for the build to work. The difference is whether they link to a C++ standard library DLL (aka MultiThreaded[Debug]DLL, or the default) or to a static C++ standard library (aka MultiThreaded[Debug]). The vast majority of device adapters link to `MMDevice-SharedRuntime`, but some device adapters can only be built by linking to a static standard library because they link to other (vendor) libraries that also do so.

nanthony21, 2021-03-16:
Thanks @marktsuchida.

closed 2021-03-16 by nanthony21