micro-manager/micro-manager #1154: Linux - UI doesn't update during live acquisition

tomhanak, 2021-03-17:
I followed build instructions in [doc/how-to-build.md](https://github.com/micro-manager/micro-manager/blob/master/doc/how-to-build.md) combined with MM wiki for 1.4 version and installed dependencies listed in [one user’s script](https://github.com/CougPhenomics/mm_scripts/blob/master/build-mm.sh) (first line with apt only), solved all build issues but the result doesn't work as expected.

I used Ubuntu 18.04 as a base. It has Java 11 as default version and Java 8 can be installed from official repo, older version are not available. No matter the Java version the behavior is the same, just different issues popped up during build.

I started the application with Demo camera only, all defaults.
When clicked Snap the Preview windows shows generated image with diagonal stripes and on place where is shown FPS value is text "new image".

Then I clicked Live. No change in Preview window observed as well as no metadata updates in Inspect window. Taking a look at Plugins -> Developer Tools -> Sequence Buffer Monitor shows the buffer is filling continuously. When changed exposure time to 1 second and the buffer size increases by one every second. It means nothing is picking up frames from that buffer, thus the consequence of no UI updates, etc.

When used basic Bean shell script that just starts the acquisition and pops up frames, the sequence buffer remains almost always empty, i.e. captured frames are consumed well.

So far I've never tested MM version 2 on Linux so not sure it ever worked. MM 1.4 on Linux and Windows nightly builds work well.

Any ideas?

nicost, 2021-03-17:
The strange thing is that I have (very) occasionally seen this behavior on Windows.  I remember that it usually resolves when restarting live mode.  My suspicion was that there are multiple threads in play, and if they are started in the wrong order, one is waiting for the other, the display does not start up correctly.  This is very difficult to trouble shoot without a system that reproducibly shows the problem (which you have!)  Can you generate a Trouble report (Hep > Report a problem), and send it to me (either using the build in send report button or by email)?  Hopefully it gives a clue where to start looking in the Java code.

tomhanak, 2021-03-18:
Unfortunately, restarting live mode doesn't help, tried that many times on desktop and laptop.

nanthony21, 2021-03-22:
I haven't been able to replicate this. Testing on Ubuntu 20.04 everything appears to work as expected.

tomhanak, 2021-03-25:
Everything seem to work in Ubuntu 16.04 and 20.04 (installed in WSL) but doesn't work in Ubuntu 18.04 (installed on iron, one desktop, one laptop).

Could it be caused by different boost version?
Unfortunately, there is no easy way to install the same boost version on all Ubuntu installations. Other packages are more or less the same...

nanthony21, 2021-03-25:
I would guess that this is something purely Java related rather than having to do with Boost. Is it possible that your 18.04 install is being built with something other than Java8?

It was a long time ago, but I think I saw something like this when I tried building with Java11 on Windows.

nicost, 2021-03-29:
Highly doubt this is a boost issue, this must be coming from Java itself.  Just to be sure, this is with the current 2.0 version from git master branch?

Scrolling through the code displaying live images, it is difficult to envision it going wrong.  The only clear thing is that the display happens on the EDT (the Swing Event Dispatch Thread), so it may be possible that the EDT hangs (for whatever reason).

It would be useful to look in the Corelog file generated by MM to see if there is anything useful there.  The live mode code is nicely contained in org.micromanager.internal.SnapLiveManager.  Going through that code with a debugger would quickly show what is going wrong in your environment.

nicost, 2021-03-30:
You may try with debug mode on (Tools > Options), with a little luck that provides some more output.

nicost, 2021-03-31:
It would be nice to know what exactly `fromPropertyMap `chokes on, but I suspect that it is the use of a comma rather than a dot in the numerical properties.  with the script:

```
mmc.snapImage();
tagged = mmc.getTaggedImage();
tagged.tags;
```

I get:

`{"Z-Description":"Demo stage driver","Camera-PixelType":"16bit","Camera-Binning":"1","Core-Shutter":"Shutter","Camera-FastImage":"0","Z-Name":"DStage","Camera-SimulateCrash":"","Emission-Name":"DWheel","ChannelIndex":0,"Camera-TransposeMirrorX":"0","Camera-TransposeMirrorY":"0","Shutter-State":"0","Camera-Mode":"Artificial Waves","Z-Position":"0.0000","Core-AutoShutter":"1","Camera-UseExposureSequences":"No","Dichroic-State":"0","Dichroic-Name":"DWheel","Path-Description":"Demo light-path driver","SliceIndex":0,"Position":"Default","CameraChannelIndex":0,"Path-Name":"DLightPath","Camera-Description":"Demo Camera Device Adapter","Dichroic-Description":"Demo filter wheel driver","Camera-ReadNoise (electrons)":"2.5000","Camera-RotateImages":"0","Dichroic-HubID":"","PixelSizeUm":1,"Camera-BitDepth":"16","Camera-DisplayImageNumber":"0","Camera-FractionOfPixelsToDropOrSaturate":"0.0020","Core-ChannelGroup":"Channel","Path-HubID":"","Excitation-Name":"DWheel","Frame":0,"Camera-OnCameraCCDYSize":"512","Core-ImageProcessor":"","Core-Camera":"Camera","Camera-CameraID":"V1.0","Objective-State":"1","XY-TransposeMirrorX":"0","XY-TransposeMirrorY":"0","XY-Name":"DXYStage","Camera-MultiROIFillValue":"0","Excitation-Description":"Demo filter wheel driver","Width":512,"Camera-SaturatePixels":"0","Autofocus-Description":"Demo auto-focus adapter","Camera-Name":"DCam","Excitation-HubID":"","Camera-CCDTemperature":"0.0000","Camera-TransposeXY":"0","Camera-Gain":"0","Autofocus-HubID":"","Shutter-HubID":"","Camera-DropPixels":"0","Camera-TestProperty1":"0.0000","Camera-TestProperty2":"0.0000","Camera-TestProperty3":"0.0000","Autofocus-Name":"DAutoFocus","Camera-TestProperty4":"0.0000","Z-UseSequences":"No","Camera-TestProperty5":"0.0000","Camera-TestProperty6":"0.0000","PixelSizeAffine":"1.0;0.0;0.0;0.0;1.0;0.0","Emission-ClosedPosition":"0","Shutter-Description":"Demo shutter driver","Core-Initialize":"1","XY-HubID":"","BitDepth":16,"Emission-State":"0","Emission-Description":"Demo filter wheel driver","Core-AutoFocus":"Autofocus","Z-HubID":"","Camera-CameraName":"DemoCamera-MultiMode","Camera-ScanMode":"1","Camera-TransposeCorrection":"0","Objective-Label":"Nikon 10X S Fluor","Core-TimeoutMs":"5000","Channel":"DAPI","Objective-HubID":"","Dichroic-ClosedPosition":"0","ROI":"0-0-512-512","Shutter-Name":"DShutter","XY-Description":"Demo XY stage driver","Slice":0,"Camera-Exposure":"10.0000","Core-Galvo":"","Camera-MaximumExposureMs":"10000.0000","Camera-Photon Conversion Factor":"1.0000","Camera-ReadoutTime":"0.0000","Dichroic-Label":"400DCLP","Emission-HubID":"","Camera-HubID":"","Camera-Photon Flux":"50.0000","Camera-TriggerDevice":"","Excitation-State":"0","Binning":"1","Core-XYStage":"XY","Path-Label":"State-0","Camera-AllowMultiROI":"0","Excitation-ClosedPosition":"0","Height":512,"PixelType":"GRAY16","Emission-Label":"Chroma-HQ620","Objective-Name":"DObjective","Excitation-Label":"Chroma-D360","Core-SLM":"","Path-State":"0","Objective-Trigger":"-","Camera-CCDTemperature RO":"0.0000","Camera-Offset":"0","Core-Focus":"Z","Camera-OnCameraCCDXSize":"512","Objective-Description":"Demo objective turret driver","Camera-StripeWidth":"1.0000","PositionIndex":0,"FrameIndex":0}`

The Java code does not expect the use of a comma as the decimal separator rather than the dot.  This clearly is a locale issue that I have never seen before.  I have been trying to chase where this may happen, and have a hard time.  The only thing that looks suspicious to me is in MMDevice/DeviceBase.h the function CreateFloatProperty on line 622 uses ostringstream for the conversion of a double into a string.  Apparently, ostringstream can take std::locale as an input.  It is possible that the default locale for your system is something different from the "C" local.  See: https://www.cplusplus.com/reference/locale/locale/.  You may try to set the locale somewhere (either systemwide or in DeviceBase.h).  It would be wonderful to fix this in the code.

tomhanak, 2021-03-31:
It's locale! When started the app this way ```LC_ALL=C ./mmimagej```, everything works.
My WSL installations use default ```C``` locale but on iron I like English language with localized numbers and other metrics, so it looks:
```
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=cs_CZ.UTF-8
LC_TIME=cs_CZ.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=cs_CZ.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=cs_CZ.UTF-8
LC_NAME=cs_CZ.UTF-8
LC_ADDRESS=cs_CZ.UTF-8
LC_TELEPHONE=cs_CZ.UTF-8
LC_MEASUREMENT=cs_CZ.UTF-8
LC_IDENTIFICATION=cs_CZ.UTF-8
LC_ALL=
```

I know it's not easy task to write an app respecting all aspects of localization, but here is something screwed up badly. When I set ```LC_NUMERIC=C``` only, float numbers from C/C++ code are converted to strings with a dot and MM works, but e.g. exposure time in main windows or most of numbers in Property Browser show decimal separator with a dot and thousands separator with comma regardless current locale.
Output with ```LC_NUMERIC=cs_CZ.UTF-8``` doesn't differ from ```LC_NUMERIC=C``` which is confusing. Using ```LC_ALL=cs_CZ.UTF-8``` finally shows numbers in Czech format, but the app doesn't work...

```CreateFloatProperty``` is just one of many places where is ```std::stringstream``` used to convert numbers with current locale. Fixing it all would take a lot of time and it would break again sooner or later.
The only way out I see is forcing ```C``` locale directly in ```mmimagej``` Bash script.

I don't have Windows with other than US locale to test.

nicost, 2021-04-15:
The Micro-Manager C++ code expect C locale (to be honest, I only discovered now that C++ has locale support in the stringstream classes).  The Java layer fully supports locales and converts accordingly, which is likely the reason you suspected things are "screwed up badly".  The best action would be to force C locale during startup of MMCore.  Any idea if that is possible/easy?

tomhanak, 2021-04-15:
I've found in ```mmstudio``` a file ```NumberUtils.java``` that formats everything according to user's locale, just double and integer numbers are converted to/from C locale that's supposed to go to/from MMCore. I'm not sure this assumption was ever correct or cleary stated for device adapter developers.

AFAIK, in C it is only possible to change locale globally (for an app or library), in C++ it can be done globally or per-stream. Since MMCore uses C++ we could change to ```C``` locale for C/C++ with following code:
```
std::locale::global(std::locale("C"));
std::cout.imbue(std::locale());
```
The ```imbue``` method should be called on all already existing streams after chaning global locale, so also on ```std::cerr``` and ```std::cin``` if used anywhere in MMCore. With their ```wchar_t``` alternatives it's even more complicated. Not talking about locale in Boost at all, hopefully nobody uses Boost for number (de)serialization (see e.g. [how to support UTF-8 encoded paths](https://stackoverflow.com/q/45572209)).

Alternatively, we could change only number formatting for current locale like this:
```
std::cout.imbue(std::locale().combine<std::numpunct<char>>(std::locale("C")));
```

However, according to specification, the ```C``` locale should be set by default for C/C++. Using user's locale must be explicitely enabled. Unfortunately, I didn't find with full-text search where that happened.

None of the above snippets changes locale used by device adapters. And because of some adapters (like PVCAM) converts some numbers to strings, then pass it to MMCore and converts strings received from MMCore back to numbers, there is no easy way out than forcing C or English locale for whole app.

But I'm not an expert in writing localized apps at all...

tomhanak, 2021-04-23:
I tried to change locale for MM on Windows 10 by adding ```-Duser.language=cs -Duser.region=CZ``` options to ```ImageJ.cfg``` file.
The formatting of all numbers like decimal and thousands separators changed accordingly in Device Property Browser, but the acquisition ran live without issues as before. But the Inspect window shows number in metadata as it was converted with English locale.

So it is not reproducible on Windows because the locale either works differently compared to Linux or changing it via CLI options isn't sufficient.

nicost, 2021-04-23:
The -D options will only change the local for the Java code, not for the C++ code (which is exactly the behavior you describe).  I was never aware of the ability to change the C++ layer locale, but I guess that if you would use these Java flags on linux that you will get the same result as on Windows.