micro-manager/micro-manager #1161: Installed Java with MM installer

henrypinkard, 2021-03-25:
I'm trying to add a headless mode to Pycro-Manager (https://github.com/micro-manager/pycro-manager/pull/262), in which you can call a function from the Python side that starts a JVM, instantiates an `MMCoreJ`, a `ZMQServer`, and an Acquisition Engine. Then this allows you to use most of the functionality of micromanager without ImageJ/studio, so you can do things like implementing your own GUI (@tlambert03 has a very cool one based on napari).

If I'm not mistaken, when you install a nightly build of MM it also comes with its own JDK/JRE. Is this correct? Is there any way I can automatically figure out which version this is, so that I can use it specifically to start the JVM and avoid any potential version mismatch errors?

nicost, 2021-03-29:
A JRE is only provided on Windows, i.e. not on the Mac.  On Windows the JRE is in the micro-manager installation folder /jre/bin/java.exe.  The complete path on my system is /c/Program\ Files/Micro-Manager-2.0gamma/jre/bin/java.exe.  Note that this does not tell you which version it is, although I guess you can figure it out by running it with the appropriate flags.

henrypinkard, 2021-03-31:
Ok, thanks for the explanation. This should be straightforward to automate on windows systems.