micro-manager/micro-manager #1199: RAMPSHub not working with Marlin 2.0 firmware

zacharynevin, 2021-04-20:
Hi,

**TLDR:**
The `ReadResponse` function in `DeviceAdapters/CNCMicroscope/RAMPSStage/RAMPS.cpp` seems to hang indefinitely with Marlin firmware. Specifically, it seems that the `ReadResponse` or `SendCommand` functions don't work properly with Marlin firmware.

**Full issue description**

I am trying to use the Marlin firmware loaded onto a RAMPS 1.4 board to control an XYZ stage with micro-manager.

I have read through the RAMPSHub code, and it seems that the way it works is by issuing standard GCODE commands that are interpreted by the firmware on the Arduino board. Based on this, I didn't think it mattered what type of GCODE-parsing firmware is being used, as long as it can parse the GCODE commands passed over Serial and send back an "ok" response. 

MM is able to detect RAMPS in the Hardware Configuration Wizard. However, there seems to be a problem with the way that responses from the Arduino are interpreted. For example, the logs for the `RAMPSHub::Busy()` command show:
```
2021-04-19T22:42:56.551837 tid0x70000903a000 [IFO,dev:RAMPSHub] busy expected OK, didn't get it.
2021-04-19T22:42:56.551844 tid0x70000903a000 [IFO,dev:RAMPSHub] ok
2021-04-19T22:42:56.555711 tid0x70000903a000 [IFO,dev:RAMPSHub] busy expected OK, didn't get it.
2021-04-19T22:42:56.555712 tid0x70000903a000 [IFO,dev:RAMPSHub] ok
```
It seems that an "ok" response is being returned correctly by the Arduino, but the RAMPSHub code is not interpreting it properly. From the logs it looks like I am getting the proper "ok" response. In addition, when I try to use the stage control, the system hangs and I get timeout errors. I feel like there is an issue with sending/receiving commands from the Arduino running Marlin firmware.

nicost, 2021-04-20:
Code for that device adapter is here: https://github.com/micro-manager/mmCoreAndDevices/blob/fee2c1e310e2d237c1f6dda412c262efaecdf094/DeviceAdapters/CNCMicroscope/RAMPSStage/RAMPS.cpp.

You can get more logging output by switching on debug log mode (Tools > Options), and even more by activating logging on the serial port (in the serial port settings in the Hardware Configuration Settings).  The device adapter receives an answer before it receives "ok", so it does not see the "ok".  I suspect that there is a line ending issue.  Perusing the source code, it looks for line ending "\n".  If your device happens to send "\n\r", then there will be trouble, like you experienced.  Probably easiest to fix on the firmware level, otherwise feel free to send us a PR, but make sure that the updated device adapter will still work with the existing firmware.  You can also try contacting the original author.

closed 2021-04-20 by nicost

zacharynevin, 2021-04-20:
Thank so much!