micro-manager/micro-manager #1235: Inconsistent DataProvider.getNextIndex with virtual or RAM dataset

mabruce, 2021-05-13:
After loading an acquisition from disk as either a virtual or RAM dataset, the DataProvider gives different results for getNextIndex(...). The virtual dataset yields 0 for missing axes, RAM dataset yields 1. 

My guess is that the virtual dataset is using StorageMultipageTiff and this line is the culprit:
https://github.com/micro-manager/micro-manager/blob/21e8bb67bda63bf36f8481cf1d3aaf3486d95dc4/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/StorageMultipageTiff.java#L842

nicost, 2021-05-13:
The correct "NextIndex()" for a missing axis is 0, so it looks to me that the problem is that the RAM dataset returns 1. It sounds like you see this differently, so please let me know before I "fix" the wrong thing;).

mabruce, 2021-05-13:
My thinking was that a missing axis is functionally equivalent to a singleton axis (e.g. a single image still has a timepoint and a position), but I see in the commit logs that missing axes are being treated as 0 for Coords comparison purposes. Your interpretation is correct. I was looking at it as a 1-1 replacement of AxisLength(), but I'm guessing NextIndex() goes deeper and would affect adding images to a datastore or some such thing.

Also, I realized I was using DataProvider.getAxisLength where SummaryMetadata.getIntendedDimensions was actually what I intended to use. Unintended bug find!

nicost, 2021-05-13:
Thanks for it though!  This was a remnant of my work to make it possible to use Coords without having to set all axes to zero.  Keep those bug reports coming!.

PR #1236 (2021-05-13): Fixes wrong value returned by getNextIndex in DefaultDatastore for missing Axes.
Closes #1235
closed 2021-05-13 by nicost