micro-manager/micro-manager #1237: Adopt a code of conduct

marktsuchida, 2021-05-17:
How does everybody feel about adopting this code of conduct, for this repo as well as the others under github.com/micro-manager?
[Contributor Covenant Code of Conduct](https://www.contributor-covenant.org/version/2/0/code_of_conduct/)

I think we can be proud of our history of constructive collaboration, but it is good to have an explicit statement.

The Contributor Covenant is one of the most widely adopted ones in open source projects, but it is by no means the only one we could use as a template. Alternative suggestions are also welcome.

nicost, 2021-05-17:
We have an Code of Conduct on futureMMCore: https://github.com/micro-manager/futureMMCore/blob/main/CodeOfConduct.md.  Is it possible to link betwen the various repositories, or should this file be copied?

marktsuchida, 2021-05-17:
Oh! Could we maybe put it in its own repository (say, mmCodeOfConduct), and link to it from every project (e.g. from readme)? Mostly to emphasize that it applies to everything (as it already states), not just 'future'.

nicost, 2021-05-17:
A repository for a single document seems a bit of overkill, but it would be nice to have it in a "project-wide" place.

@henrypinkard  and @nanthony21, what do you think?

nanthony21, 2021-05-17:
I would vote to have the code of conduct either in this repository or on the website. Then other repositories can easily provide a link to it in the readme.

marktsuchida, 2021-05-17:
An alternative to creating a new repo would be to copy the doc into the 'micro-manager' repo and link to that from all other projects. Or copy into each project. I do think it would be nice to be in a location that is central to the project, and futureMMCore is not the obvious place to look. (If futureMMCore wants to keep its own copy, that works for me.)

henrypinkard, 2021-05-17:
I agree with @nanthony21. Keep master copy in main repo and/or website. Link all others to it

nicost, 2021-05-17:
Is "main repo" micro-manager/micro-manager?

henrypinkard, 2021-05-17:
yes

nicost, 2021-05-18:
Done.

closed 2021-05-18 by nicost