micro-manager/micro-manager #1242: Stage control, fails to do 0.1 step resolution

louwrensdev, 2021-05-25:
The Stage Control of micro-manager seems to round to integer micron, steps of 0.1-0.9 micron do not seem to be possible.

I am usign Prior stage that has 0.1 um resolution. It works fine at 0.1um with their control software, but not in MM's Stage Control. 

If I try setting a 0.1 step size for X or Y and  press the Up/Down/Left/Right buttons, I see the correct position displayed. But this is only very momentarily (less than half a second?) and next the updated X Y position is shown in micron, rounded(?) to integer values.

I believe it is _setting_ the rounded position to the device, because 
 * repeatedly stepping 0.1-0.5  um will never produce a sum of such steps and it does not move to the next integer um position.
 * stepping with 0.6-0.9 um step size will effectively make it step with 1um.

I want to help to debug this, and would like to know how micromanager / imageJ is currently built from an IDE (IntelliJ?) Could you please point to instructions on how you develop/compile it?

marktsuchida, 2021-05-25:
Thanks for your offer to help! Here are instructions on setting up IntelliJ: https://micro-manager.org/wiki/Using_IntelliJ
(Unfortunately manual setup is necessary at the moment.)

Also you can confirm that it is an issue with Stage Control rather than the device adapter (C++ code for the Prior stage) by testing from the Script Panel with commands such as `mmc.setRelativeXYPosition(0.1, 0.1);`, `mmc.getXPosition();`, and mmc.getYPosition();`. If those don't work, then the problem is probably in the device adapter.

louwrensdev, 2021-05-31:
Thanks for the feedback. I will give a try to install it with those instructions.

I guess we have a problem in the device adapter. Please have a look at this script and output:
```
mm.scripter().message("Start at: "+String.valueOf(mmc.getXPosition()));
mmc.setRelativeXYPosition(-1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
mmc.setRelativeXYPosition(0.1, 0.0);
Thread.sleep(10);
mm.scripter().message("End at: "+String.valueOf(mmc.getXPosition()));
```
Example Output (running 5 times):
```
Saved file: test.bsh
bsh % 
Start at: 28346.0
bsh % 
End at: 28346.0
bsh % 
Start at: 28346.0
bsh % 
End at: 28345.0
bsh % 
Start at: 28345.0
bsh % 
End at: 28344.0
bsh % 
Start at: 28344.0
bsh % 
End at: 28343.0
bsh % 
Start at: 28343.0
bsh % 
End at: 28342.0
bsh %
```

So ``-1`` works (most of the time -- the first time it didnt),
But ``+0.1`` times 10 does not.

I keep the Stage Control window open, and I see the X position change when running this script: it shows in rapid succession, for instance: 
```
28,232
28,231
28,231.1
28,231.2
28,231.3
28,231.4
28,231.5
28,231
28,231.1
28,231.2
28,231.3
28,231.4
28,231
``` 

I guess that around the time of printing an expected ``28,231.6`` it gets the updated position by polling interrupt and it updates the display accordingly -- it didn't move although the display made it seem it did.

I was wondering if it could be that MM somewhere accidentally casts the value to integer before it gets send to the device adapter, I even tested in Java if multiplying an `int` (the `dx` value is integer somewhere) and a `double` (the step size) results in a `double`. It does :)

How do get to the source of the device adapter? because it looks like the stage never got the message to go `+0.1` maybe it got the message to go `+0` instead.

How do get to the source of this device adapter? Please let me know.

nicost, 2021-05-31:
https://github.com/micro-manager/mmCoreAndDevices/tree/main/DeviceAdapters/Prior

louwrensdev, 2021-06-01:
Thanks! I looked but have to look harder for the bug. 

I hit a hiccup when trying to compile with intellij after following the instructions (https://micro-manager.org/wiki/Using_IntelliJ):
```
...\micro-manager\mmstudio\src\main\java\org\micromanager\internal\MMStudio.java:567:26
java: no suitable constructor found for ZMQServer(java.util.Collection<java.lang.ClassLoader>,java.util.function.Function<java.lang.Class,java.lang.Object>,java.lang.String[],<anonymous java.util.function.Consumer<java.lang.String>>)
    constructor org.micromanager.internal.zmq.ZMQServer.ZMQServer() is not applicable
      (actual and formal argument lists differ in length)
    constructor org.micromanager.internal.zmq.ZMQServer.ZMQServer(java.util.Collection<java.lang.ClassLoader>,java.util.function.Function<java.lang.Class,java.lang.Object>,java.lang.String[]) is not applicable
      (actual and formal argument lists differ in length)
```

Any Idea what I'm doing wrong?

marktsuchida, 2021-06-02:
@louwrensdev It's probably the version of one of the JARs (probably `PycroManagerJava-<version>.jar`) being older than the source code you are trying to build with. If your IntelliJ settings are pointing to a downloaded Micro-Manager installation, you probably need to update the latter.

louwrensdev, 2021-06-08:
Thanks! that helped, now I can compile with Intellij. Secondly, I want to be able to compile the MMCore and friends too, but it didn't work out of the box with Visual Code. Should I follow this hint: https://forum.image.sc/t/micromanager-build-dependencies-in-github-version-and-build-instructions/52419/12
Please let me know.

On topic:

It looks like there is not a bug in the Prior device adapter after all 😄 although this snippet indicates some confusion in the code:
https://github.com/micro-manager/mmCoreAndDevices/blob/main/DeviceAdapters/Prior/Prior.cpp#L1412
```
int XYStage::GetResolution(double& resX, double& resY)
{
   // This is a little unclear in the manual, it is possible that this works on some controllers, not others
   const char* commandX="RES,s";
   //const char* commandY="RES,Y";

   int ret = GetDblParameter(commandX, resX);
   if (ret != DEVICE_OK)
      return ret;
   resY = resX;

   /*
   ret = GetDblParameter(commandY, resY);
   if (ret != DEVICE_OK)
      return ret;
   */

   return ret;
}
```

Here is a snippet from the Prior manual relevant to these commands:
[image]
Note that ``RES,s`` (the command used in the device adapter) will return the resolution for the "s" axis, which means the XY stage. ``RES,z`` is another valid command, which returns the Z stage resolution. Neither ``RES,x`` nor ``RES,y`` are valid, indeed it is not super clear from the manual.

After some testing, it seems that the device's microsteps setting (controlled via either ``SS,<microsteps>`` or ``RES,s,<microsteps/25>`` where 25 microsteps/um is a hardware property of the stage), limits the minimum travel that the stage will do: it limits travel to integer multiples of this step size. This fact is not reflected in uManager.

Also, it gets a bit noisy / flaky if setting ``SS,5`` (or setting ``RES,S,0.2``, i.e. 0.2um) step or smaller. 

The default was ``SS,25``, or ``RES,S,1``, i.e. 1um step. That is why I couldn't step at 0.1um. Also, given that the device has 25 microsteps per 1um, it is physically impossible to achieve 0.1um step. Later I found this explicitly written in the manual:
[image]

Apart from the flakiness, I can patiently try steps of 0.08um (rather than 0.1um), by using``SS,1`` or ``SS,2``.  It seems I have to tell it to move 0.12um to get a 0.08um movement. Stepping at 0.2 (or 5x0.04) is reliable though.

I'm thinking to make a change in the code of MMStudio (or the device adapter) to reflect this understanding. I doubt if it is universally applicable (I mean any stage would have a minimum step, wouldn't it?).
Maybe the StageControl dialog input fields should round to the valid stepsizes (integer multiples of the microstep).
Maybe the StageControl dialog should have a box to modify the stage microsteps value.

I see a method ``int XYStage::OnStepSizeX(MM::PropertyBase* pProp, MM::ActionType eAct)``, is this ever called with ``AfterSet`` as action type? (I peeked at other XYStage implementations, but none seem to interface to hardware at this point... Hmm. Any tips?

----

In any case, I wanted to do a raster scan with XYStage (at higher resolution that 1um -- that's why I opened this issue first).
The problem is: I do not want MM to do any stitching logic; we have a spectrometer camera and the xy pixels of it do not map to XY position. 

Can you please point me to the place were I should find or create a "raw raster" scan feature? Tell me if I should open another issue for this.

marktsuchida, 2021-06-08:
You need Visual Studio (with C++ components), not VS Code, to compile device adapters or MMCore.

The usual way to do a grid acquisition (which I think is what you mean by a raster scan) in MMStudio is to use the "Create Grid" button in the Position List to generate grid positions. Then you can run a Multi-D Acquisition; no stitching is performed. I'm not sure if this will work with your small step size requirements, though. Beyond that, you might need to resort to scripting (you can also fill the Position List from a script and run the MDA from the GUI).

The step size means different things for different stages: sometimes it is a hardware step or microstep (for stepper motor stages only); sometimes it is an arbitrary unit used by the stage controller or its communication protocol; sometimes it is an arbitrary unit used by the device adapter (because the MMDevice interface unfortunately requires it).

As for `OnStepSizeX()` in `Prior.cpp`, it is used to define a _read-only_ property called `StepSizeX_um` (you can enable display of read-only properties in the Device Property Browser), so it is never called with `AfterSet`.