micro-manager/micro-manager #1243: option to write data as N5 rather than ome.tiff

bryantChhun, 2021-05-26:
With the new N5 plugin for FIJI, we should provide a third saving option (single page tiffs, tiff stacks, and N5).

The problem is that fast acquisitions only write to `ome.tiff` in a performant way.  However, when these datasets get large, the time to load `ome.tiff` files can be substantial.  For example, a dataset with shape (8, 121, 5, 17, 2048, 2048) , (P, T, C, Z, Y, X), is roughly 600 GB and can take hours to load.

We can subsequently convert the `ome.tiff` into `zarr`, but this adds an intermediate file and processing overhead still.  Saving directly as N5 (which I understand is basically the same as zarr) would be great.

marktsuchida, 2021-05-26:
All for it; help welcome:)
This project might require first improving the Datastore API and/or make it a standalone library first.
See also @henrypinkard's [NDTiffStorage](https://github.com/micro-manager/NDTiffStorage), which adds one more to the list of potential saving options.

marktsuchida, 2021-05-28:
@bryantChhun It's not so much that I have specific issues but rather that I feel that the interfaces are too big, meaning that they require you to reinvent lots of code to implement a new datastore. I.e. the pluggability may not be at the best level. Also the interface probably makes a lot of assumptions about the underlying storage format. Whether it is more practical to just accept what is there and try to implement n5 or if it is better to try to improve the interface is a good, open question.

henrypinkard, 2021-06-02:
I think making a NDTiff Datastore adapter so it can be used outside of pycro-manager/Magellan would solve a lot of these problems without the need for N5 support. As of NDTiff2.0 you can write >1 GB/s and the huge datasets can be opened in a couple seconds. You can then open as a dask array, which gives you essentially the same functionality as zarr, with a single line of code. It wouldn't supports the parallel writes to cloud storage that N5 does though.

I plan to do this at some point in the future, though not in the next few months, so if anyone else wants to take a crack at it feel free

jondaniels, 2023-06-16:
I think there has been progress in making NDTiff widely available in MM2, right @henrypinkard ?

Related to the original topic I recently heard a short talk about standardizing on OME-Zarr as presented [this preprint](https://www.biorxiv.org/content/10.1101/2023.02.17.528834v1).  I have no idea how close to a de-facto standard OME-Zarr has become, but it seems to have an appeal for reading data.  I also don't know how hard it would be for MM2 to write directly to OME-Zarr, and/or convert easily post-acquisition. 

Hearing this talk opened up my eyes to the community brainpower being spent on FAIR-friendly image formats.  It would be a shame if the Micro-Manager community wasn't involved in the conversation.  I don't have the expertise to contribute directly but I can raise awareness here among the people who do...

henrypinkard, 2023-06-16:
Correct, it is now fully available in both Pycro-Manager and Micro-Manager.

The issue with OME NGFF in my understanding is that it is designed for reading performance not writing performance. So far as I can tell [Zarr does not yet have the features](https://forum.image.sc/t/ngff-ome-zarr-how-fast-can-you-write-it/74303/6?u=henrypinkard) needed for high performance writing of data on a single computer.

jo-mueller, 2026-08-17:
Late to the party and saw that my colleague @lubianat was already here 🙂  I would also chime in on the pitch of supporting writing ome-zarr directly from µManager. Especially with the new version [almost released now](https://ngff.openmicroscopy.org/specifications/dev/index.html), this brings a lot of nice QOL features for µManager-written data:

- ome-zarr supports sharding now, which reduces the footprint of number of file on file systems considerably
- some backends capable of reading/writing (ome-)zarr are quite fast, i.e. [tensorstore](https://github.com/google/tensorstore).
- n5 these days is also [capable of writing ome-zarr up until version 0.5](https://forum.image.sc/t/fiji-can-read-and-write-zarr3-ome-zarr-v0-5-new-n5-release/121905) and I hope we get 0.6 soon, too

Cool features for µManager brought by ome-zarr 0.6 is mainly _transformations_ and scenes: I.e., a microscope could write individual tiles on-the-fly and could then store the coordinates of every tile (i.e., in motor coordinates) directly with the data. This could make it super-easy for stitching tools to directly consume this transformation metadata and do a one-click stitching without having to parse pos-files.