micro-manager/micro-manager #1247: ChannelSettings can not be removed from DisplaySettings

nicost, 2021-06-03:
I can not find a way to remove channelSettings from DisplaySettings using the API (other than rebuilding the DisplaySettings from scratch).  

I needed this functionality as  I run into a strange bug where the DisplaySettings restored from profile contain more channels than the store, resulting in a display with extraneous channels (in this case, it suggests that there are multiple cameras in the configuration, which there are not).  @marktsuchida is there a clean way to deal with this?

marktsuchida, 2021-06-03:
Looks like an API bug. `DisplaySettings.Builder` needs a method `Builder channels(Iterable<ChannelDisplaySettings> channelSettings)`, and passing `null` or an empty iterable/collection/list should clear all channels.

There could also be, if convenient, `Builder removeChannel(int channel)` and `Builder clearChannelSettings(int channel)` -- the former would remove the channel, shift any later indexed ones forward, and decrement numberOfChannels, whereas the latter would keep the channel but revert to the state equal to having been created with `channel(int channel)`. I suggest that both or neither of these be added, because having just one might be confusing.

nicost, 2021-06-03:
Should we do that before release?  I know this goes against our API freeze, but we clearly need them to fix this (quite perplexing, I still do not understand it completely, and it is hard to reproduce) bug.

marktsuchida, 2021-06-03:
I think fixing an API bug like this is okay.

PR #1253 (2021-06-09): Display settings API change, add function to DisplaySettings.Builder
To enable setting (and deleting and replacing) all channelDisplaySettings at once.

Failed to replicate the strange bug that I had seen before of multi-camera display appearing with a single camera config file, so did not put the new function in action yet.

Also lots of CheckStyle cleanup, but that should all be inconsequential, other than that the code looks prettier and the Javadoc improved.

Closes #1247.
closed 2021-06-09 by nicost