micro-manager/micro-manager #1249: What is PerformanceMonitor doing?

nicost, 2021-06-04:
The display code is sprinkled with calls to add data to PerformanceMonitor.  However, I failed to find instructions how to actually look at the data collected by PerformanceMonitor.  I am also a bit worried that PerformanceMonitor may actually decrease performance, especially in very high speed performance.  I currently making sure that the code still works when PerformanceMonitor is null, which should mitigate this (I guess that checking if an object is null is not too costly in the scheme of things.  This seems a nice thing, but is not useful without documentation. Remove or document?

marktsuchida, 2021-06-04:
It is enabled by setting the system property `"org.micromanager.showperfmon"`. We could document that somewhere, but I don't feel that this feature is useful to anybody who doesn't already have deep knowledge of the display code -- the data it displays is uninterpretable without reading the code.

I don't think the performance impact is that big when not enabled (it should be noted that most of its uses are per display cycle, which is never over 60 Hz, I would hope, no matter how high the acquisition rate), but please feel free to remove its use entirely from the viewer classes (or add code that would cause it to be JIT-compiled to nothing when not enabled) if you prefer it that way, whether or not there is any proof. At least that will make the viewer code cleaner, and we can always temporarily re-add performance sampling when working on display performance. In my hands, the insight gained from it was quite valuable, because most of our viewer performance issues (with the exception of histogramming and metadata processing) are not CPU-bound but rather concurrency programming bugs or nonidealities, such as waiting for things incorrectly or scheduling too many things at the same time. I'm a bit worried that trying to address display performance issues without this kind of tool is like shooting in the dark. But it is true that it is something I quickly whipped up for my own use, hence the lack of documentation. If you would like, I can make it a separate library and completely remove it from MMStudio (that would maybe force me to document it...).

The only `org.micromanager.internal.utils.performance` usage that cannot be trivially removed is the `DisplayUIController`'s use of `TimeIntervalRunningQuantile` to compute a *running median* (yes, it can be done) of actual past display FPS in order to adjust its target FPS. This is something that seemed to work better (across a wide range of incoming frame rates and processing delays) than other methods I tried, but by no means is it perfect.

nicost, 2021-06-07:
How does on set the system property `"org.micromanager.showperfmon"`?

marktsuchida, 2021-06-07:
Either from code/script panel: https://docs.oracle.com/javase/8/docs/api/java/lang/System.html#setProperty-java.lang.String-java.lang.String- (`System.setProperty("org.micromanager.showperfmon", "true")`)
Or from JVM command line args: https://docs.oracle.com/javase/8/docs/technotes/tools/windows/java.html (`-Dorg.micromanager.showperfmon=true`)
I usually did the latter, I think, but the former will probably work, especially if done before opening any displays.

nicost, 2021-06-07:
Closed in PR #1252 1252

closed 2021-06-07 by nicost