micro-manager/micro-manager #1254: SerialManager fails to reopen just-"closed" port

marktsuchida, 2021-06-10:
See https://forum.image.sc/t/micromanager-serial-port-failure-when-saving-opening-configuration-file/53705 for details.

First thing to do is probably to study the codepath (including within Boost) to the `CloseHandle` call.
Also ensure there are no changes between Boost versions we use/have used.

marktsuchida, 2021-06-14:
- I could reproduce the problem with a USB-serial adaptor (VID=FTDI, PID=6001).
- Script to reproduce (run with no hardware config loaded):
```java
port0 = "COM3";
sm = "SerialManager";

void loadAndInit(p) {
	print("Connecting " + p);
	mmc.loadDevice(p, sm, p);
	mmc.initializeDevice(p);
}

void unload(p) {
	print("Disconnecting " + p);
	mmc.unloadDevice(p);
}

loadAndInit(port0);
mmc.sleep(500);
unload(port0);
loadAndInit(port0);  // Occasionally fails, CreateFileA() error 5
mmc.sleep(500);
unload(port0);
```

- The failure rate appears to be highly variable (10-100%).
- Recording some of the Win32 API calls in the failure case using [API Monitor](http://www.rohitab.com/apimonitor) shows that the serial port `HANDLE` (=`0x7c0`) is correctly closed:

```
#	Time of Day	Thread	Module	API	Return Value	Error	Duration
96	11:48:51.094 AM	1000	mmgr_dal_SerialManager.dll	CreateFileA ( "\\.\COM3", GENERIC_READ | GENERIC_WRITE, 0, NULL, OPEN_EXISTING, FILE_FLAG_OVERLAPPED, NULL )	0x00000000000007c0		0.0027282
205	11:48:51.101 AM	1002	mmgr_dal_SerialManager.dll	GetQueuedCompletionStatus ( 0x0000000000000f24, 0x00000000264ffb40, 0x00000000264ffb58, 0x00000000264ffaa0, 500 )	FALSE	258 = The wait operation timed out. 	0.5124256
206	11:48:51.614 AM	1002	mmgr_dal_SerialManager.dll	GetQueuedCompletionStatus ( 0x0000000000000f24, 0x00000000264ffb40, 0x00000000264ffb58, 0x00000000264ffaa0, 500 )	TRUE		0.1067507
209	11:48:51.720 AM	1002	mmgr_dal_SerialManager.dll	CloseHandle ( 0x00000000000007c0 )	TRUE		0.0000126
210	11:48:51.720 AM	1002	mmgr_dal_SerialManager.dll	GetQueuedCompletionStatus ( 0x0000000000000f24, 0x00000000264ffb40, 0x00000000264ffb58, 0x00000000264ffaa0, 500 )	FALSE	995 = The I/O operation has been aborted because of either a thread exit or an application request. 	0.0000072
211	11:48:51.720 AM	1002	mmgr_dal_SerialManager.dll	GetQueuedCompletionStatus ( 0x0000000000000f24, 0x00000000264ffb40, 0x00000000264ffb58, 0x00000000264ffaa0, 500 )	TRUE		0.0000017
273	11:48:51.830 AM	1000	mmgr_dal_SerialManager.dll	CloseHandle ( 0x000000000000169c )	TRUE		0.0000051
274	11:48:51.830 AM	1000	mmgr_dal_SerialManager.dll	CloseHandle ( 0x0000000000000a40 )	TRUE		0.0000028
275	11:48:51.830 AM	1000	mmgr_dal_SerialManager.dll	CloseHandle ( 0x000000000000116c )	TRUE		0.0000044
278	11:48:51.830 AM	1000	mmgr_dal_SerialManager.dll	CloseHandle ( 0x0000000000000f24 )	TRUE		0.0000023
279	11:48:51.830 AM	1000	mmgr_dal_SerialManager.dll	CreateFileA ( "\\.\COM3", GENERIC_READ | GENERIC_WRITE, 0, NULL, OPEN_EXISTING, FILE_FLAG_OVERLAPPED, NULL )	INVALID_HANDLE_VALUE	5 = Access is denied. 	0.0000168

```

- The I/O completion port `HANDLE` (=`0xf24`) is also closed before the port is opened again with `CreateFileA`. Although this happens right before the call to `CreateFileA`, it happens on the same thread, so there is no chance of overlap.
- There are 3 additional `HANDLE`s that are closed between the serial port and the completion port, probably also created by `boost::asio`, but they are also not culprits as they are closed on the same thread.
- The serial port `HANDLE` _only_ is closed on the async thread (`1002`), but there is a significant time delay after its closure before the next `CreateFileA` and the calls do not overlap.
- This is consistent with there being a race condition in the drivers or kernel. It is possible that the fact that we have an uncanceled async read when we close the port triggers this (though I haven't determined that). (_Unless_ there is some other (non-`HANDLE`) resource used by `boost::asio` that I am unaware of and that is not closed.)
- By far the easiest workaround is to wait briefly _after_ we join the async thread.

marktsuchida, 2021-06-14:
For the record, a script to test reliability:

```java
port0 = "COM3";
sm = "SerialManager";

void loadAndInit(p) {
	print("Connecting " + p);
	mmc.loadDevice(p, sm, p);
	mmc.initializeDevice(p);
}

void unload(p) {
	print("Disconnecting " + p);
	mmc.unloadDevice(p);
}

int failure = 0;
int success = -1;

for (int i = 0; i < 101; ++i) {
	try {
		loadAndInit(port0);
		++success;
	}
	catch (Exception e) {
		print(e.getMessage());
		unload(port0);
		++failure;
		--i;
		--success; // Next success doesn't count
		mmc.sleep(50);
		continue;
	}
	mmc.sleep(100);
	unload(port0);
}

print("Failed " + failure + " out of " + (failure + success) + " times");
```

closed 2021-06-14 by marktsuchida

S3LSteffen, 2021-11-22:
I still have problems with closing/reopening the port: 

- using MM nigthly build: MMSetup_64bit_2.0.1_20211115.exe
- Win10 x64 
- device: XCite Novem, connected via USB-virtual COM, drivers: USBSer.sys, hhdserial64.sys

**Closing** the COM virtual port for this device **takes ca 5 seconds**! The 1000ms timeout of  the timed_join in SerialPort::Shutdown is too short in this case!
Increasing the timeout to 6s fixes for me the problem . Your reliability scripts (thank you Mark for that!) shows bo more problems then...