micro-manager/micro-manager #1255: Really tiny GUI when running from IntelliJ windows 10

henrypinkard, 2021-06-10:
Running from source on Windows 10/IntelliJ, the GUI and ImageJ are waaay smaller than expected (see pic). This doesn't happen when running nightly build. Has anyone experienced anything similar? I'm not sure where to even begin trying to fix this...

[image]

nicost, 2021-06-10:
Lucky you, you have a high dpi monitor!

Go to the Micro-Manager folder, right click ImageJ.exe, select Properties from the bottom of the drop-down.  In the compatibility tab select "Change high DPI settings".  Check the checkbox under "Program DPI".  It is always a bit mysterious, but clicking around in that section always fixes it (keep good notes while doing it!).

I have looked into setting this right in the INNO installer but could not find it.  Have fun with Windows!

henrypinkard, 2021-06-11:
Got it, thanks!

closed 2021-06-11 by henrypinkard

fducau, 2021-06-11:
Hi! I am experiencing the same problem on Ubuntu 18.4 and MM version 2.0.x.dev. 

I managed to change the scaling for ImageJ (_Edit>Options>Appearance>GUI Scale_), but unfortunately this does not seem to be inherited by MM Studio. 

Any clues on how to scale the MM Studio GUI in this setup?

fducau, 2021-06-15:
For reference, I managed to increase the size using the `run_scaled` script from https://github.com/kaueraal/run_scaled. The resolution is not optimal since it blurs a bit the GUI but it is usable.

nicost, 2021-06-15:
I expect there is some setting you can use on Ubuntu to make this work better.  Maybe the following tread has some pointers: https://forum.image.sc/t/imagej-1-52p-on-high-dpi-screen/29716