micro-manager/micro-manager #1264: how to debug  micro-manager with camera ?

lwxGitHub123, 2021-06-23:
hi,
      we want to debug  micro-manager   with  Raptor Photonics camera, we  could build  the C++ components,such as MMCoreJ and DemoCamera ,but we do not konw how to do next, could you give me an instructions  please ?

marktsuchida, 2021-06-23:
If by "debug" you mean run with the Visual Studio debugger, then the best way is to build the `mmgr_dal_RaptorEPIX.dll` in Debug configuration, copy it to an existing Micro-Manager installation, start Micro-Manager by double-clicking, and then attach the debugger (https://docs.microsoft.com/en-us/visualstudio/debugger/attach-to-running-processes-with-the-visual-studio-debugger?view=vs-2019).

By the way, no big deal, but we encourage you to use https://forum.image.sc/ for support questions (please tag with 'micro-manager').

closed 2021-06-23 by marktsuchida

lwxGitHub123, 2021-06-25:
got it; thank you very much !!!
  

 

 ------------------ 原始邮件 ------------------

lwxGitHub123, 2021-06-25:
hi,
   I have post the questions in https://forum.image.sc/t/micro-manager-can-not-open-the-raptor-photonics-camera-and-build-rapp-failed/54391        
   Thanks
   Liu
  

 

 ------------------ 原始邮件 ------------------