micro-manager/micro-manager #1278: Extend Stage control window with Snap after Move option

nicost, 2021-07-13:
This is often the goal, i.e. move one field of view and snap.  Adding this option, could possibly retire the Beta plugin "Snap on Move Preview".  Thoughts @marktsuchida?

marktsuchida, 2021-07-13:
Do you mean moving the functionality of Snap on Move Preview to the Stage Control window? Snap on Move Preview works even if you move the stage manually (on the device).

nicost, 2021-07-13:
But that feature of "Snap On Move Preview" improves complexity quite a bit. The Snap on Move plugin is still listed as Beta, and I (and I guess many others), do not really date use it.  Adding this functionality to the Stage Control window seems simple, straight forward and easy to maintain.  If so desired, even the polling feature of "Snap On Move Preview" could be brought into Stage Control.

marktsuchida, 2021-07-14:
Not sure what the point of reinventing the polling mechanism is -- the only reason Snap on Move Preview is "beta" is because, at least at the time, I felt that polling could be dangerous (could interfere with other processes, like an acquisition). Other than that, it can be moved out of beta any time (it's beta, not experimental). But if you don't like it, please feel free to replace it with a new implementation.

@AndrewGYork (on whose request I wrote the Snap on Move Preview): could you maybe update us on how well it is working?

nicost, 2021-07-14:
No intent to re-invent polling.  The Stage Control plugin does not need it (but it could be added there). In any case, there is no point to having related functionality in disparate places, and Stage Control is already a front panel item, so it makes a lot of sense to move the Snap on Move functionality there.  Could be as straight forward as adding a checkbox.  There already is a "polling" checkbox there.  Checking both could be the current "Snap on Move" functionality, checking only "Snap after move" would only snap if one of the arrows is pressed.  The polling behavior should be paused while an acquisition is running.

marktsuchida, 2021-07-14:
Ah! Please go ahead, I'm all for fewer lines of code. I was confused because it sounded like you were newly adding polling.

The only thing that will go away is the potential to extend Snap on Move to monitor other things (like device properties) for change. But that doesn't currently exist, so nobody will miss it. That and the ability to set the sensitivity (threshold) -- although this could be added to StageControl if found necessary.

PR #1289 (2021-07-19): StageControl plugin new features
Closes #1278.
- Adds option to snap an image after moving.
- Now moves stage to move the sample as expected based on the image, rather than based on the stage coordinates.
- Lots of checkstyling the internal navigation code.
closed 2021-07-19 by nicost