micro-manager/micro-manager #1301: Version numbering going forward

marktsuchida, 2021-08-13:
How shall we do this?

The de facto standard for anything that has an API is [SemVer](https://semver.org/), which we already try to follow for the MMCore API (though that number is not seen by many). We could adopt this for MMStudio, so that e.g. the next MMStudio release can be 2.1.0 (since we've already added to the API in #1241).

But I don't think SemVer makes sense for the whole of the Micro-Manager distribution. (For one thing, we'd have to increment the major version every time the device interface version changes, no matter how minutely.) "2.0" is not a SemVer, nor was 1.4.x. We could continue by incrementing the major or minor version based on subjectively how big the changes are. Or we could switch to some variant of [CalVer](https://calver.org/) (and possibly also resume periodic releases, though that's an orthogonal matter).

In any case, I think it makes sense to decouple the MMStudio API version and the Micro-Manager distribution version, and the former really needs to be SemVer if we want to distribute via Maven.

Cc: @nicost, @nanthony21.

nanthony21, 2021-08-13:
I would agree that it makes sense to have an MMStudio API version that is independent of the Micro-Manager version just like the device interface has it's own version. 

For Python projects I have used `setuptools_scm` to do automatic versioning based on Git tags and it has worked pretty well. Maybe we could do something similar with Micro-Manager. Basically I have a `x.x.x` SemVer Git tag that I update manually on a subjective basis. Then `setuptools_scm` adds additional information to keep track of the precise Git commit that is being used. Information about the versioning scheme is here: https://github.com/pypa/setuptools_scm/#default-versioning-scheme

marktsuchida, 2023-11-01:
Superseded by #1791.

closed 2023-11-01 by marktsuchida