micro-manager/micro-manager #1308: Can we keep mmCoreAndDevices submodule synced?

marktsuchida, 2021-08-20:
Current practice:
- `micro-manager` has `mmCoreAndDevices` as submodule, but when you check out `micro-manager`, you get some old commit of `mmCoreAndDevices`.
- We therefore tell people (and ourselves, and our CI) to perform a `git submodule update --remote`, to fetch the tip of `main` of `mmCoreAndDevices`.
- The result mostly makes sense because `micro-manager` does not depend on a specific version of `mmCoreAndDevices` and it usually makes sense to use the latest.
- However, when you want to ask the question: which commit of `mmCoreAndDevices` would have been used to build a `micro-manager` commit from 2 weeks ago, there is no correct way to get an exact answer.

What I would like:
- Every time there is a commit pushed to `mmCoreAndDevices`'s `main`, we (automatically) create a commit on `micro-manager`'s `master` that bumps the submodule pointer.
- We tell people _not_ to use the `--remote` flag with `git submodule update`. Initial clones can use `git clone --recurse`.
- This way, we have complete history of `micro-manager` to `mmCoreAndDevices` correspondence.

nicost, 2021-08-20:
I would like this too.  Only downside may be that we get lots of commits in the `micro-manager` repo, but that seems a minor downside to me.

marktsuchida, 2021-08-20:
Yes, this is true: the overhead will be negligible but perhaps somewhat visually cluttering.
At least the commits will be one per push to `mmCoreAndDevices`, not one per commit.

marktsuchida, 2021-08-20:
Incidentally, @nicost, would I be correct in thinking that the mmCoreAndDevices commit that appears in this commit
https://github.com/micro-manager/micro-manager/commit/4a5d51ea76f89eaa6e74d0d772822701661aa76a
is the one that was used to build 2.0.0? (If so, it will be good to tag it `v2.0.0`).

nicost, 2021-08-24:
May be.  I made sure to commit the specific revision of mmCoreAndDevices just before the build, although I do remember that there were build issues that needed new commits to resolve.  The micro-manager repo tag 2.0 should contain the correct mmCoreAndDevices version.  Feel free to tag that repo.

nicost, 2021-08-24:
@marktsuchida, can you set up the automated commit creation in micro-manager, and update the documentation?  I would need to work my way into Github actions and have lots of other things on my plate right now.

marktsuchida, 2021-08-25:
Yeah, I'll do this soon.

nicost, 2021-09-13:
Do we have a solution?  I am a bit confused about the currently desired workflow and how to work with mmCoreAndDevices.  Clobbering along, but feel that I could easily make nasty mistakes.

marktsuchida, 2021-09-13:
The only change I made related to this PR is that now when somebody pushes to the `main` of mmCoreAndDevices, an automated commit is pushed to the `master` of micro-manager, so that in each micro-manager commit, the mmCoreAndDevices submodule points to the latest code.

Therefore, we should now use `git submodule update` without `--remote`, to preserve the version correspondence between micro-manager and mmCoreAndDevices. (`--remote` ignores the specified submodule commit and checks out the latest unconditionally.)

The only thing to be careful about is to avoid manually changing the mmCoreAndDevices version when committing to micro-manager. (So you don't want to do `git add mmCoreAndDevices`.)

For the (Jenkins) CI, the `--remote` flag cannot yet be removed, because it is needed to get the latest SecretDeviceAdapters.

As I wrote above, I propose that we change the Jenkins build to use the master/main branches, rather than `privateMain`, and manually clone the latest SecretDeviceAdapters under the `mmCoreAndDevices/` directory. Then we can delete the `privateMain` branches from both mmCoreAndDevices and micro-manager. Should simplify things.

marktsuchida, 2021-12-10:
The remaining issues have been addressed by #1358 and https://github.com/micro-manager/mmCoreAndDevices/pull/101.
(It turned out that `privateMain` was _not_ being used by the Jenkins CI.)

closed 2021-12-10 by marktsuchida