micro-manager/micro-manager #1319: Changing exposure sometimes fails

nicost, 2021-09-13:
Specifically, when a channel group and channel have been set, and are "active", i.e. the system has the Channelgroup-Channel state, changing the exposure time for certain devices (such as PVCAM cameras) fails.  It turns out that this is caused by the OnPropertiesChanged call from the PVCAM adapter, and this same problem can be reproduced by adding the OnPropertiesChanged call to the DemoCamera adapter.  

Likely related to https://github.com/micro-manager/mmCoreAndDevices/issues/31.

I am tempted to disable listening to the OnPropertiesChanged call while changing the exposure time in the MM GUI, but it is probably better to look at the underlying problems as discussed in https://github.com/micro-manager/mmCoreAndDevices/issues/31

marktsuchida, 2021-09-13:
To solve the root problem of https://github.com/micro-manager/mmCoreAndDevices/issues/31, we need something like an `OnPropertyRangeChanged` (note singular), which notifies that the limits or allowed values of a property changed. Then PVCAM (and hopefully others) won't need `OnPropertiesChanged` anymore, and we can deprecate it (and ignore it in MMStudio), or possibly remove it entirely.

So the first question is whether it is easy for mmstudio to handle such an event correctly...

nicost, 2021-09-13:
I guess that it should be straightforward to update the UI elements that are affected by such a call.  However, it seems harder to decide what to do with config presets that include the property whose range was changed.  The latter may need to be addressed at the level of the Core. What else should mmStudio do with this call?

marktsuchida, 2021-09-14:
I think the UI elements are the main ones. And now that I think about it, I think we can be fairly confident that it will be more or less the things that currently respond to `OnPropertyChanged` and/or `OnPropertiesChanged`. (If it needs to respond to a range change, it probably needs to respond to a value change as well.)

I don't think the Core needs to help with the config presets, because the only case where their behavior changes is when it is a single-property config group, but the special hack for those is entirely implemented in MMStudio (I think). In theory there could be the case where currently allowed values that are part of a config group disappear, but such behavior would probably break a lot of things and device adapters shouldn't do such things.

Also, calling `OnPropertyRangeChanged` can probably be automated and hidden from device adapters -- MMDevice base classes can call it when the device adapter changes the limits or allowed values of a property, and MMCore can filter out the calls that come during `Initialize` of the device. Let me know if you'd like me to take a stab at this part.

I assumed above that it would be good to include both limits (lo/hi, slider) and allowed values (dropdown options) in the concept of property _range_, but actually I'm not sure about the latter. Maybe it should not be allowed for a device to change the list of allowed values, because it will likely break the entire config group model. So if there are no devices currently calling `AddAllowedValue` outside of `Initialize` (or pre-init properties), maybe we should just disallow it. Need to check. In any case, we can limit the callback to `OnPropertyLimitsChanged` for now, and think about the allowed values case later.

PR #1320 (2021-09-14): Studio: fix inability to set exposure time in certain conditions.
When the device adapter calls OnPropertiesChanged in the setExposure method
(such as done by the PVCAM adapter), the exposure time would reset to the
previously known exposure time for the channel and group (when synchronization
with the MDA was requested).  This was caused by the setExposure function not
updating the value stored by AcqDialog.  It so turns out that the DefaultChannelExposureEvent
was not inheriting from ChannelExposureEvent, hence the event was never processed
(this is one of those instances where I do not like [... cut]
commit 225a28c6 (2021-09-14): Merge pull request #1320 from nicost/exposure

Studio: fix inability to set exposure time in certain conditions. Closes #1319
closed 2021-09-14 by nicost