micro-manager/micro-manager #1357: Changing user profile properties order

jacopoabramo, 2021-12-08:
Greetings everyone,

I'm copying an original [post ](https://forum.image.sc/t/changing-user-profile-properties-order/60746) I created in image.sc

I have a device adapter for a laser driver which has a series of properties arranged as follows:

```
Property,LaserDiodeDriver,Laser Label 1,405nm 150mW
Property,LaserDiodeDriver,Laser Label 2,
Property,LaserDiodeDriver,Laser Label 3,488nm 55mW
Property,LaserDiodeDriver,Laser Label 4,532nm 40mW
Property,LaserDiodeDriver,Laser Label 5,445nm 1600mW
Property,LaserDiodeDriver,Laser Label 6,638nm 1200mW
Property,LaserDiodeDriver,Laser Label 7,638nm 700mW
Property,LaserDiodeDriver,Laser Label 8,
Property,LaserDiodeDriver,Max. Laser Power 1 (%),60.4
Property,LaserDiodeDriver,Max. Laser Power 2 (%),100.0
Property,LaserDiodeDriver,Max. Laser Power 3 (%),63.7
Property,LaserDiodeDriver,Max. Laser Power 4 (%),91.9
Property,LaserDiodeDriver,Max. Laser Power 5 (%),80.5
Property,LaserDiodeDriver,Max. Laser Power 6 (%),82
Property,LaserDiodeDriver,Max. Laser Power 7 (%),1.0
Property,LaserDiodeDriver,Max. Laser Power 8 (%),1.0
Property,LaserDiodeDriver,Min. Laser Power 1 (%),39.5
Property,LaserDiodeDriver,Min. Laser Power 2 (%),0.0
Property,LaserDiodeDriver,Min. Laser Power 3 (%),51.6
Property,LaserDiodeDriver,Min. Laser Power 4 (%),82.7
Property,LaserDiodeDriver,Min. Laser Power 5 (%),52.2
Property,LaserDiodeDriver,Min. Laser Power 6 (%),54.9
Property,LaserDiodeDriver,Min. Laser Power 7 (%),0.0
Property,LaserDiodeDriver,Min. Laser Power 8 (%),0.0
```

This is an extract of a CFG file we created for a device. It’s also the order of properties in the hardware configuration wizard when creating a new configuration.

As you can see there are three different groups of properties:

one for the laser label;
one for the laser maximum power;
one for the laser minimum power;
I would like to arrange them in such a way that they are grouped as follows:

- Laser N;
- Laser N Min Power;
- Laser N Max Power;
- Laser N+1;
- Laser N+1 Min Power;
- Laser N+1 Max Power;
etc…

In the device adapter the properties are generated with a for loop as follows:

```C++
   for (int i = 0; i < NUMBER_OF_LASERS; ++i) {
      CPropertyAction* pActLaserMinPower = new CPropertyAction (this, &LaserDiodeDriver::OnLaserMinPower);
      CPropertyAction* pActLaserMaxPower = new CPropertyAction (this, &LaserDiodeDriver::OnLaserMaxPower);

      char p_name[64];

      // Laser label
      sprintf(p_name, "Laser Label %d", i+1);
      ret = CreateStringProperty(p_name, "", false, nullptr, true);

      // Laser limits
      sprintf(p_name, "Min. Laser Power %d (%%)", i+1);
      ret = CreateFloatProperty(p_name, 0.0, false, pActLaserMinPower, true);
      ret = SetPropertyLimits(p_name, 0.0, 100.0);

      sprintf(p_name, "Max. Laser Power %d (%%)", i+1);
      ret = CreateFloatProperty(p_name, 100.0, false, pActLaserMaxPower, true);
      ret = SetPropertyLimits(p_name, 0.0, 100.0);
   }
```

For some reason though - which I don’t know, the order of the properties in the configuration wizard is changed and “sorted” as presented at the beginning of the topic.

Is there some option to specify to micro-manager to NOT sort alphabetically the properties and making sure that the order in the for loop of the device adapter is respected?

Thank you!

marktsuchida, 2021-12-08:
No, unfortunately there is no way to do that. Properties are not ordered, and the Device Property Browser sorts them by name. To preserve the order in which properties are defined would require changing all code that handles properties accordingly. But that would break existing device adapters that have been written under the assumption that properties are sorted when presented.

jacopoabramo, 2021-12-08:
Thank you for your reply @marktsuchida but I don't understand the rush in closing this issue. I understand it may be a silly thing, but I would be able to have the ordering of properties as I decide. Even the fact that there are device adapters that have a dependency in the order of properties is a bit puzzling, but I don't know enough to judge. Are you planning to change this approach after the Core rework?

marktsuchida, 2021-12-08:
It's not silly, it would be quite nice to have this (and I think you are not the first to suggest this, although I can't find a past issue). The only reason I closed the issue is because I read it as a question about whether this is currently possible, rather than a suggestion to change how it works.

When/if the Core is reworked/replaced/whatever, this is something that surely ought to be considered. It is also not inconceivable to allow device adapters to opt into leaving its properties unsorted (the only downside being added complexity).

I'll reopen this to see if anybody else has comments.