micro-manager/micro-manager #1369: Update dependencies to newer versions

nicost, 2021-12-23:
We seem to be quite behind, and may want to consider the versions used for imagej, imaglib2, ome, scijava, jogamp/gluegen, and possibly others.

fabio-echegaray, 2022-02-02:
Hi @nicost, first time here, and also my first PR, so please have mercy :). Regarding jogl/gluegen, I figured out that specifying the artifacts to Ivy can solve the problem of having to manually include the jar files in a local folder.  I am not entirely sure if this kind of solution is something being avoided by the team though. Finally, I checked the last version of the libraries in Maven central and they haven't changed from the version that is in ivy.xml since 2015.  

PR for this change is #1386 . Let me know if there is anything else required for the jogl/gluegen dependencies.

fabio-echegaray, 2022-02-02:
Hi @nicost, first time here, and also my first PR, so please have mercy :). Regarding jogl/gluegen, I figured out that specifying the artifacts to Ivy can solve the problem of having to manually include the jar files in a local folder. I am not entirely sure if this kind of solution is something being avoided by the team though. Finally, I checked the last version of the libraries in Maven central and they haven't changed from the version that is in ivy.xml since 2015.

PR for this change is #1387. Let me know if there is anything else required for the jogl/gluegen dependencies.

nicost, 2023-08-29:
We are a bit more up to date now....