micro-manager/micro-manager #1392: Build improvements master plan

henrypinkard, 2022-02-08:
I think it might be useful to have one issue where we can keep track of the many improvements to the build system, so that it is easier to keep on top of it and contribute. @marktsuchida @nanthony21 @nicost, does anyone want to summarize where we are headed?

FYI, I just (somewhat painfully) figured out how to automatically build and deploy pycro-manager-Java and its libraries using Github actions. Example [here](https://github.com/micro-manager/pycro-manager/blob/main/.github/workflows/maven_deployment.yml).

marktsuchida, 2022-02-08:
Yes.

marktsuchida, 2022-02-08:
Here is an overview of the short term goals. The overarching principle is to avoid changing more than one thing about the build at a time, as much as possible.

# Recently finished:
- Moved our automated builds (Jenkins) and download server to the cloud
- Dropped support for 32-bit x86 (both Windows and macOS)
- Updated macOS build to work with current Apple compilers

# Current (almost done):
- https://github.com/micro-manager/mmCoreAndDevices/issues/40

# Next:
- Switch our C++ standard from C++03 to C++14
  On Windows, this happens automatically by virtue of building with VS2019.
  Make `-std=c++14` the default for macOS and Linux.
  After this, people can contribute code that requires modern C++.

- Make our code compatible with C++17
  This mostly means replacing `std::auto_ptr` with `std::unique_ptr`.
  Updates should be made while keeping compatibility with C++14; no need for a
  branch.

- Provide code-signed installers (Windows, macOS) #396

# Later this year

We want to eliminate the hard-to-maintain dual build system (Visual Studio + GNU Autotools) and make it easier for people to get started with parts of the code and make contributions.

Below are the desired end results, **not** in order of implementation; the actual transition plan is complex (and partly TBD); separate issue(s) will be created as we go.

- Each module (device adapter, plugin, MMCore, MMDevice, MMStudio, etc.) should live in its own Git repository. Each repo should have a single license, version numbering (which can be date-based for modules not exposing an API), and CI.

  - This better matches the reality of our development: the majority of contributors work on one or a few modules, and for them, dealing with the complexity of the full MM build is counter-productive.

- Eliminate the subversion repositories `3rdparty` and `3rdpartypublic`.

  - We shouldn't have to provide binaries for open source C++ dependencies: they should be detected on the system (Unix) or else automatically built by our build system.

  - We should simplify building modules that require closed-source C++ dependencies (vendor SDKs): the default should be to find them in default install locations. (For CI, we may place binaries in private repos, one for each dependency.)

  - Java dependencies should be obtained via Maven.

- C++ modules should use Meson as build system

  - We may want to enforce some standardization for device adapters

  - Consider Meson wrap files for limited dependency management (try to avoid Git submodules as much as possible)

- Java modules should use Maven as build system and for dependency management

- CI should run on GitHub Actions

  - It should also be a goal to keep the build system and CI well separated: the build system should not care how the CI is being run, and needs to be flexible about the build environment and source of dependencies (at least for C++). The CI, on the other hand, should strive to be as reproducible as reasonably possible.

  - Build artifacts should be published (a public Maven repository for Java; somewhere public for C++). This means that device adapter and plugin contributors can test builds without waiting for a whole MM build

- Migration to this new setup should be done module by module, not all at once

  - During the transition period, our nightly installers should be assembled by combining build artifacts from the new repos and from what remains in the aggregate repos (micro-manager and mmCoreAndDevices).

  - We need good instructions for developers. The transition from just building a single device adapter in Visual Studio GUI must be smooth.

- We also need to support the use case of installing a "full" MM installation (at least on Linux)

# Other modernization topics

- Reduce use of Boost for things that are now in the C++ standard library

- Native Apple arm64 support (the build already works; perhaps only need code signing)

marktsuchida, 2022-02-08:
I guess this also partially fits under this topic and may be a prerequisite for a sane Maven-based build for Java modules:
- https://github.com/micro-manager/micro-manager/issues/916

henrypinkard, 2022-02-09:
Awesome! I can definitely help with some of the mavenization of the Java parts, once that's on deck

marktsuchida, 2022-02-10:
Also @nicost and I were thinking that it might be good to do a 2.1 release before getting into the "Later this year" parts, once things have stabilized.

@henrypinkard That would be appreciated!

henrypinkard, 2022-02-14:
@marktsuchida What time of day are the nightly builds currently going up? I've got a breaking change between Pycromanager python and java sides, and I want to sync my pypi timing with the next nightly?

marktsuchida, 2022-02-14:
@henrypinkard The builds are currently configured to start at 6:00 AM UTC and usually take about 12 minutes, after an additional 5 minutes or so for a Windows VM to spin up.

marktsuchida, 2022-06-15:
Here is a more detailed plan for how to do the build system renovation (separation into individual Git repos and conversion to new build system) in piecemeal and with minimal disruption.

Here are the code dependency relationships in Micro-Manager:
[image]
(Dashed arrow is non-optional run-time dependency, although the single one we have should be easy to make optional.)

Previously I was thinking that we should start at the bottom of the dependency graph, namely with `MMDevice`, and work our way up. While that makes logical sense, it is complicated by the fact that, as soon as we move `MMDevice`, all of the device adapters need to start using `MMDevice` at the new location; this change would have to be made in the existing build systems, which is a pain in the neck.

I now think it is actually easier to start at the middle and try to prioritize changes that go toward completing the unfinished job of splitting `micro-manager.git` and `mmCoreAndDevices.git`, and also, for the majority of components, do the build system conversion before, rather than together with, repository splitting.

1. Move `MMCoreJ` into its own repository. This repo will, for the time being, have `mmCoreAndDevices` as a Git submodule. `MMCoreJ` should be built with Meson and GitHub Actions, have its own SemVer version number (independent of the C++ MMCore version number), and releases (containing auto-extracting binaries for all platforms) should be published to Maven Central. (This is analogous to how `pymmcore` currently works.)

2. When `MMCoreJ` is ready, we can change `mmstudio`'s build scripts to treat `MMCoreJ` just like any of the third-party Java libraries that it depends on. This severs the Java build from the C++ build. The device adapters will still be built as part of `micro-manager.git`'s Jenkins CI and packaged together. The build systems are still tied together.

3. At this point, a bunch of changes become possible in parallel:
    - `MMCore` can be put into its own repo, which will have `mmCoreAndDevices.git` as a submodule for the time being (to provide `MMDevice`). `MMCoreJ` and `pymmcore` can now have `MMCore.git` instead of `mmCoreAndDevices.git` as submodule.
    - `mmstudio` can be mavenized (without moving out of `micro-manager.git`, and with the current Ant build kept in parallel). Then, `acqEngine` and plugins can be mavenized. Once all are mavenized, we can start moving them to separate Git repos, where they will be built, versioned, and published with Maven. The `micro-manager.git` repo then contains no Java code but only the scripts to build C++ components and aggregate together with Java components into installers.
    - `MMDevice` can gain Meson build scripts, without moving out of `mmCoreAndDevices.git` just yet. `MMCore` build scripts can then be modified to use `MMDevice` as a wrap module.

4. With the Java build taken care of, we can move all remaining bits of Autotools scripts used for the C++ build into `mmCoreAndDevices.git`. `micro-manager.git` is then for packaging the distribution only. `mmCoreAndDevices.git` (which now only has `MMDevice` and the device adapters) becomes self-contained, i.e., it will be possible to build by itself on Unix as well as Windows.

5. Now we start converting each device adapter to build with Meson, each as a separate project that all just happen to live in the same Git repo. Conversion includes eliminating all dependencies on 3rdparty[public]. CI for each converted adapter should be set up using GitHub Actions (firing only when files in each directory change), although at this time the build for packaging may still happen in Jenkins. We'll need a way to build and aggregate all the device adapters, which can start out simply (just build all projects, plus the legacy build, for a given commit), but can gain whatever necessary logic (see next item).

6. In parallel to the previous step, we can start incorporating device adapters that live in their own, individual repositories. This can be piloted with one that is already in a separate repo for historic reasons (`mmdev-ThorlabsKinesis`) and/or a non-critical adapter that is not currently built (`SequenceTester`). This will include a mechanism to collect separately built device adapter (and other) "releases" into a distribution package, probably based on a manifest that lists all component versions (and which is automatically kept up to date). Once such a mechanism exists, it can be applied also to the converted adapters that still live in `mmCoreAndDevices.git`, so that they are no longer built by Jenkins.

7. When all device adapters are built with Meson and GitHub Actions, it will be easy to move `MMDevice` into its own repo, so that consumers of `MMDevice` don't need to depend on the entire `mmCoreAndDevices.git`, and so that versioning of `MMDevice` can be managed more cleanly.

8. Finally, we can move device adapters into their own Git repos (some of which will contain multiple, closely-related adapters that, e.g., share code). This attains the goal of making it easier to work on single device adapters and having independent CI (and optional versioning) for each (in a less messy manner).

I didn't include every little detail above (and many details still need to be worked out), but I think this is a workable plan. It also has the property that not very much damage is done if we stall at any step.

nicost, 2022-06-15:
Amazing Mark!  Looks like a great plan to me!

marktsuchida, 2022-06-17:
Addendum to step 3b above (Mavenization of Java modules): We (probably meaning I) need to fix #916 before this step. Fixing #916 removes the distinction between plugins and libraries, and some libraries (`libraries/Gaussian` and `libraries/Projector`) can then be merged (back) with their corresponding plugins, if desired. Such merging can happen before or after Mavenizing and splitting out the repo, but is probably easiest to do before repo splitting. In either case, we should ensure the new repo contains the Git history for both modules, with the library module (containing the bulk of the actual code) being primary.