micro-manager/micro-manager #1394: ClearVolume Jar fetch failing

marktsuchida, 2022-02-09:
From output of `ant -f buildscripts/fetchdeps.xml`:

```
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 		::          UNRESOLVED DEPENDENCIES         ::
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 		:: org.micromanager#cl-kernel-bindings;0.1.1-SNAPSHOT: not found
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 
[ivy:resolve] :::: ERRORS
[ivy:resolve] 	SERVER ERROR: Bad Gateway url=https://dl.bintray.com/clearvolume/ClearVolume/org/micromanager/cl-kernel-bindings/0.1.1-SNAPSHOT/maven-metadata.xml
[ivy:resolve] 	SERVER ERROR: Bad Gateway url=https://dl.bintray.com/clearvolume/ClearVolume/org/micromanager/cl-kernel-bindings/0.1.1-SNAPSHOT/cl-kernel-bindings-0.1.1-SNAPSHOT.pom
[ivy:resolve] 	SERVER ERROR: Bad Gateway url=https://dl.bintray.com/clearvolume/ClearVolume/org/micromanager/cl-kernel-bindings/0.1.1-SNAPSHOT/cl-kernel-bindings-0.1.1-SNAPSHOT.jar
[ivy:resolve] 	SERVER ERROR: Bad Gateway url=https://dl.bintray.com/clearcontrol/ClearControl/org/micromanager/cl-kernel-bindings/0.1.1-SNAPSHOT/maven-metadata.xml
[ivy:resolve] 	SERVER ERROR: Bad Gateway url=https://dl.bintray.com/clearcontrol/ClearControl/org/micromanager/cl-kernel-bindings/0.1.1-SNAPSHOT/cl-kernel-bindings-0.1.1-SNAPSHOT.pom
[ivy:resolve] 	SERVER ERROR: Bad Gateway url=https://dl.bintray.com/clearcontrol/ClearControl/org/micromanager/cl-kernel-bindings/0.1.1-SNAPSHOT/cl-kernel-bindings-0.1.1-SNAPSHOT.jar
```
(Note that this error is not visible if the Jars are cached from past runs in `~/.ivy2`.)

nicost, 2022-02-09:
dl.bintray is no longer functional.  ClearControl seems to have moved to maven (but the version we were using is not available there).

marktsuchida, 2022-02-09:
It looks like the package in question belongs to `org.micromanager`?
https://github.com/micro-manager/micro-manager/blob/3caf8c486e668d8c3db2573f7a144e4d19c3e514/buildscripts/ivy.xml#L95-L96

nicost, 2022-02-09:
Yikes.  Looks like I put these on bintray under org.micromanager.  I guess we may put them on maven, but it may be better to see if the packages maintained by Loic (now on maven) work for us.

marktsuchida, 2022-02-10:
@nicost Where is `cl-kernel-bindings` used in Micro-Manager? (Or could it be that it is no longer used?)

marktsuchida, 2022-02-11:
Turns out it is not used. I'll remove it.

PR #1399 (2022-02-11): Remove unused cl-kernel-bindings dependency
Closes #1394.
With this change, all dependencies are resolved.
closed 2022-02-11 by marktsuchida