micro-manager/micro-manager #1396: Save function does not split large stacks intelligently

dickinson-lab, 2022-02-10:
We have a workflow that involves acquiring 10,000 frames of data, each with 3 fluorescence channels.  Because the stack is larger than 4 GB, it has to be split into multiple .ome.tif files.  However, this splitting seems to be done based on individual 2d images, rather than timepoints or channels.  For example, depending on the exact size of the image, the first file in the dataset might contain frames 1-187 of all three channels, plus channel 1 of frame 188.  The second file then begins with channels 2-3 of frame 188, etc.  

This is a problem because none of the resulting images contains a full complement of channels for each time point it contains. As a result, the individual images cannot be opened, de-interleaved and displayed correctly by any downstream software, including ImageJ/FIJI (the only exception is MicroManager itself, which opens and reconstructs the entire dataset). 

The same issue applies to images with multiple z planes, where (for example) one file might contain half the stack at a given time point, while the rest of the stack is found in the next timepoint.

Possible solutions: 
- Implement a save function that splits the data into files in some principled way: For example, require that no z stack or channel series be split between two files (and reduce the file size as necessary to ensure that this happens).  
- Allow a user (or the callling function of `Store.save()`) to specify the maximum size of the file that can be saved, thereby allowing the user to calculate and set a limit below 4GB that will result in large files being split "evenly" into tiffs that are individually openable. 

I have a slight preference for the second option because it requires fewer up-front assumptions about the nature of the user's dataset, but thoughts and suggestions are welcome.  I'd be glad to help code a solution if the experts can point me to the corner of the source code that actually decides how to split images during the saving (I looked but couldn't find it).

marktsuchida, 2022-02-11:
Indeed this is a major problem. Related: #889.

Your second option is an interesting idea, because it seems much less daunting than trying to split logically by image coordinate (which may need extending the file format, and in any case requires a fundamental reworking of the saving code). There is a chance it could be unstable if a byte limit is used, though, because the size of frame metadata may vary. Perhaps it could be made to limit the number of planes per file.

Everything having to do with saving in the multi-page TIFF format is in the package [org.micromanager.data.internal.multipagetiff](https://github.com/micro-manager/micro-manager/tree/main/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff).

The set of files split due to size limit is managed by the `FileSet` class (there is one `FileSet` per XY position, or per entire dataset, depending on the Options setting), and its `writeImage` method contains the logic to decide whether to split:
https://github.com/micro-manager/micro-manager/blob/fe1d1f77b214ed90883b41eb8369167649c07429/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/FileSet.java#L142-L168

marktsuchida, 2022-02-11:
I think the OME metadata is now saved as an external file when splitting, though I would need to check.

dickinson-lab, 2022-02-11:
Howdy Mark, thanks for the pointers on where to find the relevant code.  I will look into it and see if I can code any possible solutions.  

I thought of a third potential solution, which would be to split the data during acquisition, rather than during saving.  This might be even easier to implement.  The logic would go like this: 
Before acquiring each timepoint: 
- Calculate the size of the next frame, based on the image size and number of channels and z planes to be acquired. 
- Check whether adding another frame will make the dataset >4GB.  
- If so, create a new datastore for the next frame (and subsequent frames).
- Then, when the datastores are saved, they will automatically be in different files.  

Thoughts and feedback on this option are very much welcome.  As Henry points out, this might entail a performance cost.  Although I doubt the metadata would be a problem, because the vast majority of metadata is plane-specific (rather than associated with the whole dataset) anyway. 

@henrypinkard, we haven't tried the bio-formats importer in FIJI but it's possible it would work.  We often want to open and work with single files, though - for example in MATLAB.  And it's convenient to be able to open single files even in FIJI sometimes, since the entire datasets can be huge.

marktsuchida, 2022-02-12:
I'm not so sure about your 3rd option, because the acquisition is not in a position to determine what happens to the datastore (not should it). Even if you did it downstream of the acquisition, you would need to introduce an entirely new layer upstream of the datastore, and it would need to interpret and modify metadata to convert the dataset into separate datasets. To do that right, you would essentially need to do the same thing as option 1 (or possibly more work than option 1).

Henry was referring to the OME-XML metadata, which is separate from the Micro-Manager (per-plane and summary) metadata, and is generated after all planes have been stored. It is a single large chunk of XML, and is copied into every `.ome.tif` file that is a member of the dataset. I thought this was no longer the case with MM 2.0, but apparently it still is (just did a very quick check). Perhaps this would need to be addressed before allowing to split by size or image count.

dickinson-lab, 2022-02-12:
Thanks for the feedback, that makes sense.  Will look at the saving code next week and update you with my thoughts/progress.

henrypinkard, 2022-02-14:
> @henrypinkard, we haven't tried the bio-formats importer in FIJI but it's possible it would work. We often want to open and work with single files, though - for example in MATLAB. And it's convenient to be able to open single files even in FIJI sometimes, since the entire datasets can be huge.

If you check the "virtual" option in bioformats it wont bring the data into RAM so it doesn't matter how big it is. You can also use this option to split out individual timepoints/channels/etc

dickinson-lab, 2022-02-16:
Here's an update on my thought process so far based on digging into the code.  First, to summarize my understanding of how saving and file splitting works (please let me know if I've misunderstood anything): 
- Saving is handled at the top level by [org/micromanager/data/internal/DefaultDataSaver.java](https://github.com/micro-manager/micro-manager/blob/c0fe182e57294ab687868e49f144a32f959ba4a4/mmstudio/src/main/java/org/micromanager/data/internal/DefaultDataSaver.java).  When `store.save()` is called, a new DefaultDataSaver is created and the Datastore is passed to it.  
- The DefaultDataSaver, in turn, creates a MultiPageTIFF Datastore called `duplicate_` and fills it with the images in the Datastore it receives. Images are added to this dataset using `duplicate_.putImage` on line 153 of DefaultDataSaver.java.
- When the MultiPageTIFF Datastore receives an image as a result of a `putImage(Img img)` call, it automatically tries to write it to disk by calling `writeImage(image)` on [line 343 of StorageMultiPageTIFF.java](https://github.com/micro-manager/micro-manager/blob/c0fe182e57294ab687868e49f144a32f959ba4a4/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/StorageMultipageTiff.java#L343). 
- writeImage() then decides whether to split the data and generate a new file, as Mark pointed out above. 

The problem with this architecture is that writeImage() doesn't (and can't) know how big the whole datastore is, because it only gets a single 2D image plane.  It is therefore not in a position to split the datastore into files in an intelligent way; all it can do is decide whether the image is has will fit into the file it has, and make a new file if not.  

Instead, I think that DefaultDataSaver should decide whether and how the datastore needs to be split up before saving. DefaultDataSaver.java already contains some code (lines 100-149) to sort the axes in the right order, so it knows what axes are present and can easily estimate the total size of the dataset.  So, then my two ideas from above become: 

Idea 1 (New saving approch):  DefaultDataSaver calculates the size of a single frame (x * y * c * z * bytes_per_pixel + metadata) and decides how many frames will fit in a single file without splitting a frame. Then, instead of copying all of the data to a single MultiPageTIFF Datastore (`duplicate_`), it copies one file's worth of data to each of several MultiPageTIFF Datastores, each of which will then be saved automatically as a single file by the existing code.
Pros: More deterministic, since data are split up in a pre-defined way before we ever write anything to disk.  Also probably requires only changing code in one place (DefaultDataSaver), with the rest of the package left untouched.
Cons: Maybe less flexible (?), requires making extra Datastores, maybe a little more work to code. 

Idea 2 (Variable file size limit):  DefaultDataSaver calculates the size of a single frame (x * y * c * z * bytes_per_pixel + metadata) and decides how many frames will fit in a single file without splitting a frame.  Then, it sets a max file size and passes it as a parameter when creating the MultiPageTIFF Datastore (replacing the [current hard-coded value](https://github.com/micro-manager/micro-manager/blob/c0fe182e57294ab687868e49f144a32f959ba4a4/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/MultipageTiffWriter.java#L74-L75) of MAX_FILE_SIZE). 
Pros: Maybe more intuitive, and straightforward to modify later if requirements change. Preserves the current architecture where `writeImage()` does the actual splitting.
Cons: Requires calculating precisely how big a dataset and its metadata will be; if we get it wrong, images will end up in the wrong file.  Requires more changes across more parts of the code to create the max_file_size parameter and get it passed through to the function that needs it.  

Altogether I lean towards option 1, but would welcome your feedback.  Once you agree to an approach, I am happy to take a stab at modifying the code, or happy to have you do it if you think it's safer that way.

marktsuchida, 2022-02-17:
Thanks. I think you'll need to find a way that only involves modifying `StorageMultipageTiff` (and the classes it depends on, in the `multipagetiff` package). That is the only way to make the code work the same way in all cases (e.g. acquiring to disk vs saving after acquiring to memory). The `DefaultDataSaver` class, for example, is only used when saving (or re-saving) an existing datastore.

I was attracted with your original idea precisely because it could be implemented without knowing the details of the incoming dataset, i.e., without making architectural changes. I was imagining something where `FileSet` would have just one additional parameter: max number of planes per file. The API (probably the `Datastore` interface or the methods that create datastore objects) would need a new method to set that parameter. That way, the problem of knowing how to split files is decoupled from the problem of actually splitting files as they are saved. In any case, this is clearly a hack, but one that might be helpful if we don't promise too much intelligence.

Alternatively (and I only thought of this now), at least when the `axisOrder` and `intendedDimensions` fields are set in the summary metadata (not always the case, but should be the case with MDA), it might be possible to cause the multipage TIFF datastore to "split on the Nth axis" at the `FileSet` level.

Also, we can't ignore the problem that @henrypinkard pointed out of the duplicated OME-XML metadata. I think solving this is a prerequisite for this kind of splitting to be usable.

Sorry it's not easy....

dickinson-lab, 2022-02-17:
> Thanks. I think you'll need to find a way that only involves modifying StorageMultipageTiff (and the classes it depends on, in the multipagetiff package). That is the only way to make the code work [...]

Ah, ok, I didn't realize there were other DataSavers.  Let me see if I can figure out a way to do this. 

> Alternatively (and I only thought of this now), at least when the axisOrder and intendedDimensions fields are set in the summary metadata (not always the case, but should be the case with MDA),

These fields offer a potential solution to the problem I mentioned in my last post, by possibly allowing StorageMultiPageTiff to know how big the datastore is (it doesn't know that currently, because DefaultFileSaver only passes it one plane at a time). I will explore that next. 

> it might be possible to cause the multipage TIFF datastore to "split on the Nth axis" at the FileSet level.
> 
> Also, we can't ignore the problem that @henrypinkard pointed out of the duplicated OME-XML metadata. I think solving this is a prerequisite for this kind of splitting to be usable.

To be clear, I am not imagining each time point in its own separate file, nor am I expecting that the number of splits will increase compared to what is happening currently.  To illustrate the behavior I am aiming for, imagine I have a 50-frame time series with 12 Z planes.  Suppose current architecture puts all of frames 1-33, and half of frame 34, into one file; the rest of frame 34 and the remaining frames are in the second file.  I aim to make it put frames 1-33 into the first file, then split early so that the entirety of frame 34 goes with the subsequent frames into the second file. Either way there are two files (and two copies of the metadata).  But the resulting files will now be openable individually because they each contain a full complement of planes at each timepoint.  If I can manage to do this, I don't think the metadata duplication that @henrypinkard mentioned would be an issue.

marktsuchida, 2022-02-17:
Yes, but once the feature exists, other people will use it in ways that it splits into more files, and they are bound to complain that the total disk space increases.... For example, somebody doing a time-position-Z-channel acquisition (in that axis order) may decide they want a new file for each time-position, perhaps because their analysis runs at the unit of multi-channel Z stacks. But I see your point: it is not a prerequisite but rather a parallel problem, so no need for you to solve it!

dickinson-lab, 2022-02-22:
Thanks again for all your suggestions, Mark.  I was able to figure out a solution that is pretty simple.  There was only one change needed.  The `hasSpaceToWrite` function of `MultipageTiffWriter` queries the `axisOrder` and `intendedDimensions` in the summary metadata, and uses this information to calculate the expected size of a complete set of channels and slices (instead of just calculating the size of a single plane, as before).  If the full frame can't be accommodated in the available space, `hasSpaceToWrite` returns false, triggering `FileSet` to split the data as before. 

I tested this with all possible combinations of boxes checked in the MDA window and found that it performed as expected (including producing the desired splitting behavior) in every case.  It should also work for custom acquisition plugins as long as these fields are set.  If they are not set, it defaults to the old behavior. 

Let me know if you think any more changes are needed, but personally I am happy with this solution.

dickinson-lab, 2022-03-16:
I made updates to my [earlier pull request](https://github.com/micro-manager/micro-manager/pull/1415) that addressed your questions - just posting here to ask you to take a look in case you didn't see the update already.  I'm eager to get these changes merged so we can update and start to make use of the improved save functionality.

closed 2022-04-01 by marktsuchida