micro-manager/micro-manager #1402: Consider bundling JRE with macOS installer

marktsuchida, 2022-02-14:
Suggested by @henrypinkard in #396, for consistency with Windows installer.

It may increase download size (currently ~92 MiB) by 40-50 MiB and install size by ~125 MiB.

Apple ARM64 builds of Java 8 JRE are not available from Adoptium but are available from Zulu.

tlambert03, 2022-02-17:
If this is done, would be possible to have a second distribution without the additional JRE/size?  I think it makes sense to have this be the standard offering, but for those who don't want the JRE, it would be nice not to have to get it

marktsuchida, 2022-02-17:
@tlambert03 Would that be useful on Windows, too?

tlambert03, 2022-02-19:
I would like that!  But, to the original point of this post: I'd say that consistency between mac and windows is probably top priority.  So, if you want to distribute only installer for each platform, they should probably all contain java as that will certainly be the most common use case.  But, if it's not too much trouble, an additional slim package with just compiled C code would be great.

marktsuchida, 2022-02-19:
@tlambert03 Makes sense, thanks.

@henrypinkard I hope to get to this eventually, but it might take some time (probably not the highest priority task at the moment). In the meantime, for pycro-manager headless mode, you might be able to simplify your instructions (or even further automate your code) by using the macOS command `/usr/libexec/java_home`, which can autodetect installed JDKs and JREs:
```
$ /usr/libexec/java_home
/Library/Java/JavaVirtualMachines/temurin-17.jdk/Contents/Home
$ /usr/libexec/java_home -v 1.8
/Library/Java/JavaVirtualMachines/zulu-8.jdk/Contents/Home
```
The JRE will need to be installed the normal way for this to work, not using PyPI or Homebrew packages. Type `man java_home` for further details.

marktsuchida, 2025-06-20:
Reopening because of #2147.

marktsuchida, 2025-06-20:
BTW regarding installer size: our installer is down to ~77 MiB because unnecessary JARs were removed recently; adding the JRE takes it back up to ~107 MiB. We might be able to shave off some more (and a lot in decompression time) by switching to LZMA (instead of the current bzip2) for the DMG -- which requires macOS 10.15, but we'll likely have that requirement sooner or later just to fully allow C++17.