micro-manager/micro-manager #1404: Possible improvements to nightly build

marktsuchida, 2022-02-14:
We currently perform nightly builds by kicking off independent Windows and macOS builds at 6:00 UTC, each building the tip of `main`. This has these downsides:

- A new build is performed even if no code changes were made in the past day
    - This consumes compute and storage resources (a minor problem)
    - It also results in more work when testing past nightly builds to determine when a bug was introduced
- There is no strictly correct mapping between the code revisions (Git SHAs) and the build; it can only be estimated
- There is no guarantee that the Windows and macOS builds from the same date use the same code revisions

After recent changes (#1365 and https://github.com/micro-manager/mmCoreAndDevices/pull/101; also see #1308), we have achieved a state where the Git SHA of the `micro-manager` repo fully specifies all code used to build Micro-Manager. (This does not mean the build is fully reproducible, but at least it is unambiguous what revision of our code is used, provided that the build is set up correctly.)

It would be nice if we could make it so that it is easy to tell the exact code used for nightly builds, and also avoid performing redundant builds when the code is exactly the same.

One way to do this might be as follows:
- Create a long-lasting branch called `nightly`, which tracks `main`, but is only updated, automatically, once a day at 6:00 UTC,
- Set the nightly builds to run when `nightly` has new commits (instead of blindly at 6:00 UTC), and
- Record the Git SHA into the build as part of the version number, so that Help > About Micro-Manager can display the exact commit used in the build. (Consider also adding an API method that returns the SHA.)

(A variant of this approach would be to use tags, such as `nightly-20220214`, instead of the `nightly` branch. But that would result in a lot of tags. It does have the advantage of providing the reverse mapping from code revisions to build dates.)

For anybody looking to download the nightly builds, the only change will be that there won't be a build for every date (which is already the case due to occasional failed builds). I doubt this would disrupt anybody's workflow.

An alternative way might be to do away with "nightly builds" entirely, instead providing "development builds", which could be produced every time code is pushed to `main`. However, I don't feel this is particularly useful to non-developer users -- we would have to add not only the date but also the time to keep the builds in order, and it just makes the whole situation more complex. (We can separately provide builds for every push to `main` for the purpose of developers testing their code; these need not be kept for the long term, e.g. expiring after 14 days.)

Thoughts and feedback appreciated!

henrypinkard, 2022-02-15:
> One way to do this might be as follows:
>
>     * Create a long-lasting branch called `nightly`, which tracks `main`, but is only updated, automatically, once a day at 6:00 UTC,
>
>     * Set the nightly [...]

This seems like a great idea to me. I think it would simplify the process of non-developers downloading and trying different builds to find when bugs were introduced a lot.

> (A variant of this approach would be to use tags, such as `nightly-20220214`, instead of the `nightly` branch. But that would result in a lot of tags. It does have the advantage of providing the reverse mapping from code revisions to build dates.)

This seems more complicated to me. I'm not sure it is worth it for just that advantage. And that information could be stored by a separate mechanism anyway.

> An alternative way might be to do away with "nightly builds" entirely, instead providing "development builds", which could be produced every time code is pushed to `main`. However, I don't feel this [...]

I think from the developer perspective, its easier to just rebuild locally than to have to constantly PR and trigger rebuilds, and I even if I wanted to roll back to something from a few days ago, I'd rather just do that in git and rebuild locally. 

But it is really nice to have an easy mechanism for users to download updated versions as soon I make a change. I currently do this occasionally by having them download the .jar from the maven repo manually and drop it into the plugins folder. But this would be much simpler if they could just download and install a full, up-to-date build.

So I wonder if build could then be divided into:
1) Official releases
2) The (no more often than) nightly builds you suggest
3) A single **Latest** build, that gets triggered every time main changes, or one of the dependencies that we maintain sends a signal that it has updated

marktsuchida, 2022-02-15:
@henrypinkard Thanks for the comments. I agree that the mapping from SHA to build can be recorded elsewhere, if needed.

> I think from the developer perspective, its easier to just rebuild locally than to have to constantly PR and trigger rebuilds, and even if I wanted to roll back to something from a few days ago, I'd rather just do that in git and rebuild locally.

Of course developers should check locally before submitting changes, but local builds don't always produce the same result as the official build, because e.g. the developer forgot to add or update a dependency (especially if a new module is being added). It would be less of a problem if it were easy for everybody to do a full local build, but that is not possible, and never will be under the current scheme (another of the motivations to go to individual repositories for modules -- so that each repo's build can be more fully tested locally, as well as by PR builds).

> 1. Official releases
> 2. The (no more often than) nightly builds you suggest
> 3. A single Latest build, that gets triggered every time main changes, or one of the dependencies that we maintain sends a signal that it has updated

Well, that is pretty much what I was suggesting, except that I don't think a single "latest" build is useful. We need the `main` builds to be listed by commit hash, or else it could change in the midst of testing (which would be very confusing if collaborating remotely). And since it cannot be predicted when a new build will happen, we need to have at least the last few.

Also, I think it is very important that it is stressed that the `main` builds are only for temporary testing, not for production use (because they will no longer be available to reproduce issues later). I wouldn't even link to them from the download page; better to link to them from one of the development pages. And the name should be something like "test builds", not "latest" which might suggest the newest version for general use.

Hopefully in the future the CI builds of individual repos will cover the use case of having particular users or collaborators download the latest version of a given module for testing.

marktsuchida, 2022-02-26:
Starting today, 1.4 nightly builds are run only when there is an svn commit.

marktsuchida, 2022-02-26:
All nightly builds now build only when there has been a commit to `main` since the previous nightly build. However, they do not use a `nightly` branch yet and do not record the Git SHA yet.
A note has been added to the download file listing pages.