micro-manager/micro-manager #1409: Changing acquisition properties during live is broken

tomhanak, 2022-02-18:
It looks like a regression in MM 2.0, because everything works as expected MM 1.4 nightly build (tested 1.4.24_20220217).

Steps to reproduce:
1. Run MM 2.0 with any PVCAM camera and open Device Property Browser.
2. Start live acquisition by pressing Live button on main window.
3. In the main window change binning or exposure time.
   - Expected: The acquisition stops for a while, reconfigures with new settings and starts again.
   - Actual result: Works as expected.
4. Change the settings back but via Device Property Browser.
   - Expected: The acquisition stops for a while, reconfigures with new settings and starts again.
   - Actual result: The acquisition doesn't stop, the new property value is applied according to core log, but in the GUI it looks like it is rejected and previous value is shown.
5. Click the Refresh button on Device Property Browser in order to fix the problem from step 4.
   - Expected: The acquisition stops for a while, reconfigures with new settings and starts again.
   - Actual result: Works as expected on some computers, but on others the acquisition soft-hangs (must be stopped and started again via buttons).
6. Stop the acquisition and start it again via buttons on main window.
   - This always applies the most recent settings and updates Property Browser view.

It is kind of reproducible with Demo camera as well. Unfortunately, the Demo camera doesn't allow change of binning during live (shows an error popup). But when the exposure time is changed, the nightly builds 20191003 and older accept new settings and reconfigure the acquisition. Newer builds cause a hard-hang, there are no new frames coming and when pressed Stop button, whole GUI stops responding and the process must be killed via Task Manager.

I have tracked down the last working nightly build is 2.0.0-gamma1_20191003.
The pull request micro-manager#73 could be the root cause, esp. the changes from [nanthony21/propertyBrowserDoubleUpdate branch](https://github.com/nanthony21/micro-manager/commit/48561a741c49b48f41dc78bd4ad4fcbb3c919510).

marktsuchida, 2022-02-18:
Thanks for tracking down the nightly build.

Marginally related: #537.

nicost, 2022-03-12:
The DemoCamera hang was caused by the code calculating and emulated exposure time in such a way that increases in the exposure time caused an extra wait (depending on the number of images in the buffer and the increase in exposure time).  That problem is fixed in https://github.com/micro-manager/mmCoreAndDevices/pull/178.