micro-manager/micro-manager #1418: State Device that declares large number of positions results in unwieldy number of Labels

nicost, 2022-02-28:
Currently, when a state device creates a "State" property with 64 states, for instance using the following code:

```
pAct = new CPropertyAction(this, &DigitalOutputPort::OnState);
CreateIntegerProperty("State", 0, false, pAct);
SetPropertyLimits("State", 0, 63); 
```

the Hardware Configuration Wizard will create 64 labels (org.micromanager.interbal.hcwizrd.LabelsPage), that can be edited by the user.

Labels were meant for things like filter wheels, where each position has a filter with a known name.  However, with digital output ports, labels are rarely useful, and the list of 64 (unused) labels in configuration files with Arduino  have always bothered me.

The Hardware Configuration Wizard can be "tricked" in not creating labels by  returning zero in the State Device API function "GetNumberOfPositions", however doing so certainly is a hack, and any code relying on that function will produce erroneous results.

On the other hand, for a 32 pin digital output port with 2^32 states, the HCW will use up all available memory in the JVM, so letting it create these labels is not workable.  I could declare the device as a generic device, but then will miss out on very useful utilities like the StateDeviceShutter.  

Are there clean ways within the existing API to convince the HCW not to generate labels?  If not, is there a minimal extension to the API that would let us do this cleanly?

marktsuchida, 2022-02-28:
This was the reason why I artificially limited `DATTLStateDevice` to 8 DA devices, even though it could otherwise apply to more channels.

We can imagine adding a conditional to the HCW to avoid creating labels when there are greater than a certain number (say, 256, to avoid changing behavior for most existing devices) of allowed values (with no change to the API).

If that still creates `Label` lines when resaving the config, maybe we could avoid writing `Label` lines when they are only repeating the default.

There may be an additional need for the Device Property Browser to avoid dropdowns when there are too many options. And the config group editor when creating a single-property group on `Label`.

And any other (e.g. plugin) code that enumerates labels, or enumerates all choices of multiple-choice properties, has the potential to break.

So the reality is that we cannot safely have more than a reasonable number of states, if we care about not breaking things. At least not without going through a lot of trouble.

If there is any way to avoid having a greater-than-8-bit `State` in your device (I presume NIDAQ), I would strongly consider those options.

nicost, 2022-02-28:
Yes indeed, this is for a NIDAQ device with a 32 pin output port.  Presenting this as 3 x 8 pin ports would be a bit of a nightmare, and restricting to 8 pins just because of this issue also feels wrong.  Lying about the number of states seems to work well, at least there are no issues with the Device Property Browser (it only seems to look at the property "State", which has correct Property Limits), or with the Group Presets.

The Device Adapter I am writing does not have a "Label" property defined, so the only interaction with Labels is through the API.  Would it make sense for the HCW to check if a Label property is defined, and only offer setting labels when there is a "Label" property?

I also like your suggestion to only define labels that are different from the proposed "State-#" ones.  Happy to implement those two changes.

marktsuchida, 2022-02-28:
Oh, I like that. It didn't occur to me that you could get away with not defining the `Label` property (I guess the property creation itself is left to the device adapter even though the implementation is in `StateDeviceBase`).

I think your plan makes sense. Am I right that you won't need to lie about the number of states once the two changes are made?

nicost, 2022-02-28:
I believe so, but will need to test.

nicost, 2022-03-01:
PR #1419 makes the HCW only show State device labels for State devices that implement the Label Property.  The State device can advertise the correct number of states without problems.  In addition, it only writes labels that differ from` State-#` to the config file.

closed 2022-03-01 by nicost