micro-manager/micro-manager #1425: Compiling on linux fails

ianhi, 2022-03-18:
With this error:

```
MMCoreJ.compile:
 [mm-javac] Compiling 29 source files to /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/build
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:231: error: incompatible types: Long cannot be converted to int
 [mm-javac]          Rectangle r = new Rectangle((int) xs.get(i), (int) ys.get(i),
 [mm-javac]                                                  ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:231: error: incompatible types: Long cannot be converted to int
 [mm-javac]          Rectangle r = new Rectangle((int) xs.get(i), (int) ys.get(i),
 [mm-javac]                                                                   ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:232: error: incompatible types: Long cannot be converted to int
 [mm-javac]                (int) widths.get(i), (int) heights.get(i));
 [mm-javac]                                ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:232: error: incompatible types: Long cannot be converted to int
 [mm-javac]                (int) widths.get(i), (int) heights.get(i));
 [mm-javac]                                                      ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:248: error: incompatible types: int cannot be converted to Long
 [mm-javac]          xs.add(r.x);
 [mm-javac]                  ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:249: error: incompatible types: int cannot be converted to Long
 [mm-javac]          ys.add(r.y);
 [mm-javac]                  ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:250: error: incompatible types: int cannot be converted to Long
 [mm-javac]          widths.add(r.width);
 [mm-javac]                      ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:251: error: incompatible types: int cannot be converted to Long
 [mm-javac]          heights.add(r.height);
 [mm-javac]                       ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/StrVector.java:73: warning: [cast] redundant cast to int
 [mm-javac] 			String strs[] = new String[(int) size()];
 [mm-javac] 			                           ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/LongVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Integer ints[] = new Integer[(int) size()];
 [mm-javac]                                       ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/DoubleVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Double ints[] = new Double[(int) size()];
 [mm-javac]                                     ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/UnsignedVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Long ints[] = new Long[(int) size()];
 [mm-javac]                                 ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CharVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Character ints[] = new Character[(int) size()];
 [mm-javac]                                           ^
 [mm-javac] /home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/StrMap.java:66: warning: [cast] redundant cast to String
 [mm-javac]     Iterator itr = find((String) key);
 [mm-javac]                         ^
 [mm-javac] Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
 [mm-javac] 8 errors
 [mm-javac] 7 warnings

BUILD FAILED
/home/ian/Documents/oss/micro/micro-manager/mmCoreAndDevices/MMCoreJ_wrap/build.xml:15: Compile failed; see the compiler error output for details.
```

ianhi, 2022-03-18:
I'm using `SWIG Version 4.0.2` if that matters

ianhi, 2022-03-18:
oh indeed it does: https://github.com/micro-manager/mmCoreAndDevices/issues/37#issuecomment-892878161

marktsuchida, 2022-03-21:
Yep. Sorry but you need SWIG 3.x on Linux and macOS.

I included quick instructions in https://github.com/micro-manager/micro-manager/blob/main/doc/how-to-build.md.

I'll leave this issue open so that it's easier to find the info when trying to build from this repo, but let's track details in https://github.com/micro-manager/mmCoreAndDevices/issues/37.