micro-manager/micro-manager #1427: Multi monitor setup prevents start on Linux

ianhi, 2022-03-18:
When I have just my laptop screen micromanager launches without issue. However, if I plug in an external monitor then launch fails with this error:

```
2022-03-18T12:13:14.514086 tid140471726065216 [IFO,LogManager] Enabled logging to stderr
2022-03-18T12:13:14.534474 tid140471726065216 [IFO,LogManager] Enabled primary log file /tmp/CoreLog20220318T121314_pid367771.txt
2022-03-18T12:13:14.534588 tid140471726065216 [IFO,App] User: ian
2022-03-18T12:13:14.534601 tid140471726065216 [IFO,App] Host: fermented-beans
2022-03-18T12:13:14.534615 tid140471726065216 [IFO,App] MM Studio version: 2.0.x dev
2022-03-18T12:13:14.534632 tid140471726065216 [IFO,App] MMCore version 10.2.0
2022-03-18T12:13:14.534642 tid140471726065216 [IFO,App] Device API version 70, Module API version 10
2022-03-18T12:13:14.534649 tid140471726065216 [IFO,App] Operating System: Linux (amd64) 5.15.23-76051523-generic
2022-03-18T12:13:14.534656 tid140471726065216 [IFO,App] JVM: OpenJDK 64-Bit Server VM, version 17.0.2, 64-bit
2022-03-18T12:13:14.536561 tid140471726065216 [IFO,App] Finished waiting for plugins to load
2022-03-18T12:13:14.762240 tid140471726065216 [IFO,App] Thread 1 (main) terminated with uncaught exception
2022-03-18T12:13:14.762510 tid140471726065216 [IFO,App] java.lang.NullPointerException: Cannot invoke "java.awt.GraphicsConfiguration.getBounds()" because "config" is null
2022-03-18T12:13:14.762558 tid140471726065216 [IFO,App]   at org.micromanager.internal.dialogs.IntroDlg.<init>(IntroDlg.java:216)
2022-03-18T12:13:14.762566 tid140471726065216 [IFO,App]   at org.micromanager.internal.MMStudio.<init>(MMStudio.java:370)
2022-03-18T12:13:14.762570 tid140471726065216 [IFO,App]   at org.micromanager.internal.MMStudio.main(MMStudio.java:195)
```

which is coming from this code: 
https://github.com/micro-manager/micro-manager/blob/47298d8df3a3c8013a900e5d7274a8c7e46948a7/mmstudio/src/main/java/org/micromanager/internal/dialogs/IntroDlg.java#L215-L216

Interestingly if I start it with just my laptop open, then plug in an external monitor it seems to work fine.

ianhi, 2022-03-18:
I Suspect part of the core of the issue is somewhere here:

https://github.com/micro-manager/micro-manager/blob/47298d8df3a3c8013a900e5d7274a8c7e46948a7/mmstudio/src/main/java/org/micromanager/internal/utils/GUIUtils.java#L214-L243

nicost, 2022-03-18:
Please try again with JDK 8.  I ran a similar setup yesterday on Windows and had no issues plugging an external monitor in, so if the problem persists, exact steps to reproduce the problem will be essential.  Or, if you are setup to debug Java code, put some break points in the relevant code and see what is going on.

ianhi, 2022-03-18:
annoyingly difficult to get jdk-8 to be used it seems. even after `sudo apt install openjdk-8-jre openjdk-8-jdk` and `sudo apt remove openjdk-17-jdk openjdk-17-jre` it was still defaulting to version 17 :/ took some conda trickery.

Do I also need to recompile with jdk8?

ianhi, 2022-03-18:
recompiled with jdk8. Still failing with multiple monitors

ianhi, 2022-03-18:
(pymm) ian@fermented-beans:~/Documents/oss/micro/micro-manager$ java -version
openjdk version "1.8.0_312"
OpenJDK Runtime Environment (Zulu 8.58.0.13-CA-linux64) (build 1.8.0_312-b07)
OpenJDK 64-Bit Server VM (Zulu 8.58.0.13-CA-linux64) (build 25.312-b07, mixed mode)

nicost, 2022-03-18:
If you compiled with JDK8 as the target, you should not have to recompile.  I have a multiple monitor setup (both monitors same size) where MM runs without issues.  The window handling code is going through quite a few hoops to handle multiple monitors (of different sizes) gracefully and seems to work OK on Mac and Windows.  Without being able to step through that code in a debugger, I am not quite sure what to do.

You could try to change line 216:

`GraphicsConfiguration best = null; `

to 

`GraphicsConfiguration best = getConfigs().get(0);`

and see what happens.  Not pretty.  I assume that the real problem is that - for whatever reason - none of the GraphicsEnvironments in your configuration intersect with 0,0.  For the fun of it you could try changing that to 1,1 (that would be a classic;).

RileyMShea, 2022-07-05:
> none of the GraphicsEnvironments in your configuration intersect with 0,0

That was the issue on Ubuntu Linux for me.

In my case, I have one monitor that is smaller than the other two(1680 vs 1920), and that had a position offset of 30 in nvidia settings:
[image]

### Workaround

I changed the position to have a vertical offset of zero and that allowed `getGraphicsConfiguration` to find the hardcoded x:0,y:0 configuration, which fixed the crash at launch.

---
I don't know what code-based solution works best to handle this(if any), but I guess it should be noted that there's a reasonable probability that any linux machine with multiple monitors(or perhaps even just one) may have a non-zero position offset on what would otherwise be the x:0,y:0 monitor, and that will crash micro-manager at launch.

marktsuchida, 2022-07-06:
Thank you, @RileyMShea, for the helpful description.

I think a proper fix could be done by getting all the `GraphicsConfiguration`s and finding the one whose top-left is closest to (0, 0).