micro-manager/micro-manager #1428: Stage stepsize illegible on linux

ianhi, 2022-03-18:
[image]

nicost, 2022-03-18:
Please try again with JDK8. To be honest, happy to see how well the code is doing with JDK17;)

ianhi, 2022-03-18:
ah indeed that helps quite a lot. Doesn't fully fix though unfortunately. Some cutting off:
[image]

nicost, 2022-03-19:
Swing is pretty  good at creating cross-platform UIs, but it is hard to get thing to look good on all.  We have often gone back and forth between Mac and Windows to settle on something that looks OK in both.  No warranties that it will be OK on another OS.  The code creating these JTextFields is on lines 518 and 519 of org/micromanager/internal/dialogs/StageControlFrame.java.  I have the feeling that the Look&Feel on your system is creating a large border.  There is some discussion of borders here: https://stackoverflow.com/questions/2281937/swing-jtextfield-how-to-remove-the-border.  You can play around with that code to see how it affects things.  Very difficult to do without feedback.  On Windows, the text is left aligned, very close to the edge, and there is no cutting off problem.