micro-manager/micro-manager #1434: Freezes after choose config screen

kobayashikirschvink, 2022-04-11:
Hi all,

I've been unable to get past the "choose config window" when I launch the program.
I couldn't load the MMconfig_demo.cfg so it doesn't seem to be the microscope's problem.
I end up with just the ImageJ window and nothing else. I attached the CoreLog file and screenshot below.

@ianhi was able to load the same config files when he did it from pymmcore-plus so this seems to be an issue on the java side rather than the C++ core.

Thanks!

[image]

[corelog.txt](https://github.com/micro-manager/micro-manager/files/8467440/corelog.txt)

nicost, 2022-04-11:
Which OS is this?  Which MM version?  Can you check the Corelogs directory and post the relevant section of the latest Corelog (if any)?

kobayashikirschvink, 2022-04-11:
I am using Windows 10, and MM version is 2.0.1 20220401. I have also attached the full Corelog in my first post below the screenshot (it says "Core log click me").

Thanks!

```
2022-04-11T17:18:09.043385 tid8436 [IFO,LogManager] Enabled primary log file C:\Program Files\Micro-Manager-2.0\CoreLogs\CoreLog20220411T171809_pid9172.txt
2022-04-11T17:18:09.043385 tid8436 [IFO,App] User: Raman
2022-04-11T17:18:09.043385 tid8436 [IFO,App] Host: DESKTOP-LD09UV5
2022-04-11T17:18:09.043385 tid8436 [IFO,App] MM Studio version: 2.0.1 20220401
2022-04-11T17:18:09.043385 tid8436 [IFO,App] MMCore version 10.2.0
2022-04-11T17:18:09.043385 tid8436 [IFO,App] Device API version 70, Module API version 10
2022-04-11T17:18:09.043385 tid8436 [IFO,App] Operating System: Windows 10 (amd64) 10.0
2022-04-11T17:18:09.043385 tid8436 [IFO,App] JVM: Java HotSpot(TM) 64-Bit Server VM, version 1.8.0_152, 64-bit
```

ianhi, 2022-04-11:
Hi Nico,

The core log is there in the expandable details block at the bottom. The short of it is windows 10 and we saw this even after updating to the latest nightly build. 

One maybe important piece of information is that there was a power outage on Saturday, which may explain why this worked last week but not this week

nicost, 2022-04-11:
Ah!  i believe that the user Profile is corrupt.  You can test this by creating a new profile in the "Startup Configuration" window.  If that works, delete the faulty profile.

ianhi, 2022-04-12:
@nicost that seems to have done the trick thank you! 

Would it be possible to catch error messages like this (I imagine in some restore window state method?) and display an error message suggesting somethign like that?

nicost, 2022-05-20:
I looked at that, and it is very complicated.  Deleting for now, but feel free to open a new issue with something along the lines of "Need better error reporting when profile is corrupt."

closed 2022-05-20 by nicost